MCUdude/MicroCore #99: Software i2c library?

MCUdude, 2020-01-03:
@nerdralph has previously mentioned that he's been working on a software based, tiny i2c library that will run on ATtiny13. It's called [picoI2C](https://github.com/nerdralph/picoI2C). Does there exist any examples on how to read and write to a device?

Will we see i2c slave support? The T13 would be a great i2c slave for gathering sensor data. But master support is most important. i2c clock frequency is also a thing. Many devices only supports a clock speed of 400 kHz, and some sone only 100 kHz. Would it be possible to implement a `setClock(uint16_t kHz)` function? Static delays should be OK for slow clocks.

It would be great if it could be wrapped into the familiar Wire-style library in the future!

nerdralph, 2020-01-03:
It's working for basic bus scans, but needs more debugging for full write/read transactions.  I recently tested it with a SSD1306 OLED, and am getting a garbled display.

MCUdude, 2020-01-03:
>  I recently tested it with a SSD1306 OLED, and am getting a garbled display.

Wow, an SSD1306 on a T13? That can't be! 😮 I was planning to test it on an AT42QT1070 capacitive touch chip I have. It has a very simple interface and is actually pretty good at what it does!

Speaking of the AT42QT1070. Can it be that this chip really is just an ATtiny84 or ATtiny841 with a disabled SPI interface. All the pins matches!

nerdralph, 2020-01-04:
I managed to hack https://github.com/datacute/Tiny4kOLED to use my I2C.  It took some extra work because it uses the Print base class.  It was around 800 bytes compiled.  Sometime in the future I'll probably do an optimized Print class with a much smaller printNumber method.
I may end up doing a fork of Tiny4KOLED, since it has some room to be further optimized.
As for the AT42QT1070, I never heard of it before.  I may take a look at the datasheet if I get a chance.

MCUdude, 2021-09-10:
I'm closing this because there now is a software i2c library available. I'd love to have i2c slave support as well, but I currently don't have the time and knowledge to do this at the moment.

closed 2021-09-10 by MCUdude

maxint-rd, 2023-08-29:
FYI: For the past years I've been working on an I2C slave library for the ATtiny13A implemented in C/C++ (no assembly skills). Of course it only supports limited features, but I've got it working for the best part. When clocked at 9.6MHz it is capable of I2C speeds up to 100kHz.  The bare minimum sketch currently takes 482 bytes of flash and 8 bytes RAM, 
I'm still testing the library while trying to add some useful examples (such as a 4-channel ADC with changeable address in EEPROM). Cramming things into just 1 kB is part of the fun...

Big THANK YOU @MCUdude for providing th wonderful MicroCore!

Sidenote: I found this [excellent example](https://github.com/wagiminator/ATtiny13-TinyOLEDdemo) by Stefan Wagner, demonstrating an I2C OLED display. I don't think he uses any I2C library, but it's very nice to see that text sine-scrolling along.

usr97629238, 2024-03-15:
> FYI: For the past years I've been working on an I2C slave library for the ATtiny13A implemented in C/C++ 

Sorry, but care to share the WIP as is?

maxint-rd, 2024-03-16:
> Sorry, but care to share the WIP as is?
Okay, I will do that. As soon as I have published something on Github, I will post a link here. (Whatever I upload will be work-in-progress, probably quite messy and containing lots of commented code, with plenty issues and todo's). To be continued...