MCUdude/MicroCore #100: delayMicroseconds() is not working

MikeX7, 2020-01-10:
Hello,

when I attempt to use the delayMicroseconds() function in my code I get the following compile time error:

```
x:\apps\arduino-1.8.10\hardware\tools\avr\avr\include\util\delay.h: In function 'main':

x:\apps\arduino-1.8.10\hardware\tools\avr\avr\include\util\delay.h:276:28: error: __builtin_avr_delay_cycles expects a compile time integer constant

  __builtin_avr_delay_cycles(__ticks_dc);

```

It works fine on Uno, so it only appears to be broken in the microcore.
Is it a bug, or did I miss some setting somewhere?

Thanks

nerdralph, 2020-01-10:
It's one of the limitations of MicroCore that delay() and delayMicroseconds() only work with fixed compile-time constants.  I've thought about changing this and submitting a pull request, but for now you can do a workaround.  Instead of delayMicroseconds(period), try:
'while (period--) delayMicroseconds(1)'

MikeX7, 2020-01-10:
Alright, when I use an integer directly, instead of a var, it works normally, so good enough for me. Thanks

closed 2020-01-10 by MikeX7