MCUdude/MicroCore #101: UART Serial library

DominicanMS, 2020-01-19:
Good evening, do you want to know if the library works just like Ralph Doncaster with just one ping? modify the core_settings.h file. 

`#define UART_TX_PIN PB0`
`#define UART_RX_PIN PB1`

MCUdude, 2020-01-19:
I don know if I understand your question, but if you only want to use Tx, for instance, it will not initialize the Rx pin if you don't call any of the Serial read functions.

DominicanMS, 2020-01-19:
AVR half-duplex software UART supporting single pin operation

[image]

MCUdude, 2020-01-19:
@nerdralph I guess one can use the setup in the picture above if both Rx and Tx pin is defined to the same pin?

nerdralph, 2020-01-19:
No, the version used by MicroCore actively drives the Tx line high.  The single-pin version uses open-drain signalling, so puts the line to high-Z, and and it is pulled up externally for high.
It might make sense to use the open-drain version since it will also work fine when Tx and Rx use separate pins, as long as the Tx line is pulled high externally.  Many USB-ttl dongles have a pullup on the Rx line for this reason.  My PL2303HX adapters have a very weak pullup on Rx, and so need an external pullup for open-drain.    I recently tested a CP2101, and they have a strong Rx pullup.

DominicanMS, 2020-01-19:
nerdralph Well, I thought it was a bad connection using the BasicSerial3 library and it works correctly

MCUdude, 2020-01-19:
@nerdralph would it be possible to implement it and trigger it by a macro in core_settings.h?

```
#define SINGLE_WIRE_SERIAL
```

Or even better, define both pins to the same physical pin:

```
#define UART_TX_PIN PB0
#define UART_RX_PIN PB0
```

nerdralph, 2020-01-20:
I wrote the half duplex code 6 years ago.  I'm pretty sure I can do better
now.  I've already started rewriting my latest BBuart code to use open
drain.  I'll make sure it works whether UART_TX and UART_RX are the same or
different.  While I'm at it, I'll improve RxByte so that it's not busy
during the full frame.  That'll give the user code more time to process the
incoming data.

nerdralph, 2020-01-21:
While working on the updated UART code, I think I figured out a way to support lower baud rates (as low as 4800bps at 9.6Mhz).  It would require some tricky coding to recognize the low baud rates, and conditionally compile a different version of the bit delay code.  What I'm wondering is how useful this would be.  If 95% of MicroCore users don't have a need for anything slower than 19.2kbps @9.6M and 9600@4.8Mhz, then I probably won't bother.

MCUdude, 2020-01-24:
> While working on the updated UART code, I think I figured out a way to support lower baud rates (as low as 4800bps at 9.6Mhz). It would require some tricky coding to recognize the low baud rates, [...]

It would absolutely great to have a wider range of supported baud rates for a given clock frequency! So your "new" code will support this plus single-wire RX/TX? I'm more than willing to test all this if you publish your code somewhere!

nerdralph, 2020-01-24:
> 
> 
> > While working on the updated UART code, I think I figured out a way to support lower baud rates (as low as 4800bps at 9.6Mhz). It would require some tricky coding to recognize the low baud rates, and conditionally compile a different version of the bit delay code. What I'm wondering is how useful this would be. If 95% of MicroCore users don't have a need for anything slower than 19.2kbps @9.6M and [9600@4.8Mhz](mailto:9600@4.8Mhz), then I probably won't bother.
> 
> It would absolutely great to have a wider range of supported baud rates for a given clock frequency! So your "new" code will support this plus single-wire RX/TX? I'm more than willing to test all this if you publish your code somewhere!

OK.  As you wish.
The earlier versions I wrote were in pure asm because I couldn't make much sense of gcc's inline asm rules.  Now that I've had a few years to figure it out, I'm writing the new version in inline asm, which should make it easier to integrate with other C/C++ code.  I should have something ready for testing in a few days.

nerdralph, 2020-01-25:
I just finished testing transmit at 9600, 38.4k, and 115.2k.  It looks good on the scope, and with a couple different USB-ttl adapters.  No docs yet, so I pushed my test code (pu.c) as well.
https://github.com/nerdralph/nerdralph/tree/master/picoUART

MCUdude, 2020-01-25:
That's great! The existing (MicroCore) code works with higher baud rates (for 9.6 MHz). Does this code too?

table from core_settings.h
```
// | Clock & baud | 460800 | 250000 | 230400 | 115200   | 57600    | 38400 | 19200    | 9600     | 4800 | 2400 | 1200 |
// |--------------|--------|--------|--------|----------|----------|-------|----------|----------|------|------|------|
// | 20 MHz       | X      | X      | X      | X (def.) | X        |       |          |          |      |      |      |
// | 16 MHz       | X      | X      | X      | X (def.) | X        | X     |          |          |      |      |      |
// | 12 MHz       |        | X      | X      | X (def.) | X        | X     |          |          |      |      |      |
// | 9.6 MHz      |        | X      | X      | X (def.) | X        | X     |          |          |      |      |      |
// | 8 MHz        |        | X      | X      | X (def.) | X        | X     |          |          |      |      |      |
// | 4.8 MHz      |        |        |        | X        | X (def.) | X     | X        | X        |      |      |      |
// | 1.2 MHz      |        |        |        |          |          | X     | X (def.) | X        | X    | X    |      |
// | 1 MHz        |        |        |        |          |          | X     | X (def.) | X        | X    | X    |      |
// | 600 kHz      |        |        |        |          |          |       | X        | X (def.) | X    | X    | X    |
// | 128 kHz      |        |        |        |          |          |       |          |          |      |      |      |

```

nerdralph, 2020-01-25:
This version is even better.  From the comments in picoUART.h:
  * picoUART is accurate to the cycle (+- 0.5 cycle error)
  * 0.64% error at 115.2k/8M and 0.4% error at 115.2k/9.6M

It'll even go to 460.8k@9.6M.  460.8k is 20.833 cycles/bit at 9.6M, so
picoUART will output 1 bit every 21 cycles, making it 0.8% slow.

nerdralph, 2020-01-25:
I just tested 500k & 921.6k on a slow t13 that runs at 9.4M instead of 9.6, with F_CPU=9400000.
At 921.6, one bit is transmitted every 10 cycles vs 10.2 for optimal, making it 2% fast.
No errors receiving with a ch330N.
For a t13 running at an exact 9.6M, 921.6k will be 4.2% fast, and therefore not recommended.

nerdralph, 2020-02-04:
The new version is ready for testing.
https://github.com/nerdralph/nerdralph/tree/master/picoUART

MCUdude, 2020-02-04:
Great! I tested your test program, from 230400 and down to 300 baud. All worked perfectly!

I'd like to try wrapping this around Serial. I understand that `pu_rx()` is blocking and returns the received character. Similar to `Serial.read()`.

Will it be enough/efficient enough if I replace `write()` with `pu_tx()`?

It seems that `prints_P()` is simply a function that prints from progmem. I was able to make use of F() instead of PSTR() to keep Arduino compatibility.

MCUdude, 2020-02-04:
Another thing; how could I make the [ReadASCIIstring](https://github.com/MCUdude/MicroCore/tree/master/avr/libraries/Serial_exampes/examples/ReadASCIIString) example to work with your code, since pu_rx() is blocking? 

Would it be possible to implement [these functions](https://github.com/MCUdude/MicroCore/blob/2952fecb276099b0b27c2b8dfe7463b064e205bd/avr/cores/microcore/HalfDuplexSerial.S#L94-L115) based on picoUART?

MCUdude, 2020-02-04:
I've just pushed some work to the [picoUART branch](https://github.com/MCUdude/MicroCore/tree/picoUART). I had to declare _pu_tx and _pu_rx as weak because the linker was complaining if not.

BTW the oscillator calibration sketch will need some work, because we don't have `TXDELAYCOUNT` and `BIT_CYCLES` anymore.

nerdralph, 2020-02-04:
> 
> 
> Great! I tested your test program, from 230400 and down to 300 baud. All worked perfectly!
Of course. :-)

> I'd like to try wrapping this around Serial. I understand that `pu_rx()` is blocking and returns the received character. Similar to `Serial.read()`.
> 
> Will it be enough/efficient enough if I replace `write()` with `pu_tx()`?
Calling pu_tx(c) in Serial.write, and pu_rx() in Serial.read() should suffice.

> It seems that `prints_P()` is simply a function that prints from progmem. I was able to make use of F() instead of PSTR() to keep Arduino compatibility.

PSTR() works fine with the Arduino core too.  I did it that way so the same code works with Arduino or just with avr-gcc/avr-libc.

nerdralph, 2020-02-04:
> Another thing; how could I make the [ReadASCIIstring](https://github.com/MCUdude/MicroCore/tree/master/avr/libraries/Serial_exampes/examples/ReadASCIIString) example to work with your code, since [...]

I'd want to look into the Arduino Serial code before duplicating the mods that James did.  There may be a better/cleaner way.  I'd also like to get a better idea of how Serial is used with existing code.  I've seen other code that uses Serial.read() and Serial.available(), I don't really know what are the most important Serial functions to get working the way they do on the standard Arduino core.  The Serial methods I'd consider a priority are those that are in common use, and not those that are rarely used outside of some example code.  Do you have any data on which Serial methods are the important ones?  A quick github search for "Serial.println" brought up 23 repositories, while a search for "Serial.available" found 5.

nerdralph, 2020-02-04:
> 
> 
> I've just pushed some work to the [picoUART branch](https://github.com/MCUdude/MicroCore/tree/picoUART). I had to declare _pu_tx and _pu_rx as weak because the linker was complaining if not.

I'd suggest looking into that a bit closer.  I intentionally did not make them weak because picoUART.h should only be included once across the build.  If it's included in different places, with and without overriding the baud rate or pin assignments, the implementation chosen by the linker may not be the one you really wanted.

> BTW the oscillator calibration sketch will need some work, because we don't have `TXDELAYCOUNT` and `BIT_CYCLES` anymore.

I used different names so picoUART can be used as a library with cores that use my earlier soft uart code.  I'll take a look at the code to see what has to change for picoUART.

nerdralph, 2020-02-05:
> I used different names so picoUART can be used as a library with cores that use my earlier soft uart code. I'll take a look at the code to see what has to change for picoUART.

Try this:
https://gist.github.com/nerdralph/a0d29e88d0c1664779d024c91ded373f
I only tested that it compiles when picoUART.h is included.  I didn't take the time to look at the picoUART branch to see where you have it included.  If it doesn't work, I'll pull the MicroCore picoUART branch and get it working with that.

Note that I reduced the cal_counter threshold to 5.  Changing OSCCAL by 1 results in a 1.3-1.5% change in frequency, so a change of much more than 3 likely means an error/bug.  Even if you were running a t13 outside, on a cold winter night, in Karasjok , I'm willing to bet OSCCAL won't need to change by more than 4.

MCUdude, 2020-02-05:
>  I don't really know what are the most important Serial functions to get working the way they do on the standard Arduino core. 

IMO the most important functionality is by far Serial.print/println. In many cases this is sufficient. Further down the list comes Serial.read() and then Serial.available(). I understand this might be difficult to implement, but a 1-byte RX buffer that's interrupt-based (INT0 and/or PCINT) where the buffer is read (and emptied) by running Serial.read() would be magic! With this functionality, I don't see much need for James' functions. However, I do understand if this would be difficult to implement.

> Try this:
https://gist.github.com/nerdralph/a0d29e88d0c1664779d024c91ded373f
I only tested that it compiles when picoUART.h is included. I didn't take the time to look at the picoUART branch to see where you have it included. If it doesn't work, I'll pull the MicroCore picoUART branch and get it working with that.

Thanks! I'll try it when I have some hardware and time available. The cal_counter variable will increment by one if the "new" OSCCAL value didn't change compared to the old one. My idea was that if the calculated OSCCAL value hadn't changed after 10 received `x` characters, the value is probably as accurate as it can get and is stored to EEPROM.

nerdralph, 2020-02-05:
> IMO the most important functionality is by far Serial.print/println. In many cases this is sufficient. Further down the list comes Serial.read() and then Serial.available(). I understand this might [...]

You should already see a big improvement in in the usability of Serial.read() since the new version exits before the stop bit (and waits for it in the next read() instead).  This gives much more time for processing the data from Serial.read().

I think I'll leave the interrupt-based receive code for a later project.  It's not based on how hard it would be to implement, but there's a couple other things that are higher priorities.  Finishing up picoI2C, and getting a final v3 picoboot release out are near the top of the list.  I'm also thinking of taking another stab at optimizing Serial.print().  In addition to solving the devirtualization optimization problem, getting rid of the support for printing arbitrary bases and just going with int to decimal and hex would save a lot of flash as well.

MCUdude, 2020-02-05:
I did try your updated OscillatorCalibration sketch with the MicroCore PicoUART fork. Sadly, it didn't work very well... Here's the output for 9.6 MHz and 115200 baud:
```
x
⸮D⸮⸮ta:`0x⸮⸮  ⸮ew ⸮⸮l: px5B
⸮lta: 0⸮⸮x  ⸮⸮w ⸮⸮l: px5B
⸮⸮⸮⸮⸮⸮: 0⸮⸮⸮ @⸮⸮⸮`⸮⸮⸮: 0⸮u⸮
⸮⸮⸮⸮: p⸮⸮⸮@`⸮⸮⸮ c⸮⸮z`p⸮u⸮
⸮⸮⸮⸮⸮⸮z@p⸮⸮⸮`@⸮⸮⸮@⸮⸮⸮z@`⸮uy
⸮⸮⸮⸮z@`⸮⸮p@@⸮⸮⸮`⸮⸮⸮z@`⸮ey⸮⸮⸮⸮⸮⸮z@p⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮ux
⸮⸮⸮⸮z@p⸮⸮e@@⸮⸮⸮@⸮⸮⸮z@p⸮ex⸮⸮⸮⸮⸮⸮z@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮mw⸮⸮⸮⸮z@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@p⸮ug⸮⸮⸮⸮⸮⸮z@p⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@p⸮mf⸮⸮⸮⸮z@`⸮⸮x@@⸮⸮⸮@⸮⸮⸮z@p⸮ef⸮⸮⸮⸮⸮⸮z@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮me⸮⸮⸮⸮r@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮r@`⸮em⸮⸮⸮⸮⸮⸮r@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮md⸮⸮⸮⸮r@`⸮⸮c@@⸮⸮⸮@⸮⸮⸮r@`⸮ml⸮⸮⸮⸮⸮⸮r@`⸮⸮⸮@@⸮⸮⸮@⸮⸮⸮r@`⸮mg⸮⸮⸮⸮r@`⸮⸮n@@⸮⸮⸮@⸮⸮⸮r@`⸮ec
```

Another thing. I tried to hardcode the OSCCAL value to a good know one to see how printing the text worked out. It seems like it tries to calculate twice every time I send an x character. This is the output from only one sent character:

```
x
DDelta: 0xFF  New cal: 0x60
elta: 0xB9  New cal: 0x60
```

> I think I'll leave the interrupt-based receive code for a later project. It's not based on how hard it would be to implement, but there's a couple other things that are higher priorities. Finishing [...]

I understand. This would be an entirely new project. However, porting James's functions would be nice.

nerdralph, 2020-02-06:
I haven't pulled the picoUART branch yet but I have started looking at the code.  It looks like you removed the naked attribute from _pu_tx and _pu_rx, which will probably mess some things up.  You can still keep them weak, at least until I have a better solution.
I've also noticed at least one circular include: Arduino.h includes HalfDuplexSerial.h, which, includes Arduino.h.  It's hard not to get distracted with all the things I'd fix/improve if I were writing an Arduino core...
With my current schedule it may be a couple more days before I have a chance to pull the picoUART branch and do some testing on the calibration sketch, so I figured I'd mention the things I noticed so far.

nerdralph, 2020-02-06:
> 
> I understand. This would be an entirely new project. However, porting James's functions would be nice.

It's not out of the question, it's just more work for me than it may seem.  While technically I could do a quick port and test it with a few hours work, I generally don't work that way.  I try to understand the full API and code path, so I can get things not just working, but working well and efficiently.

nerdralph, 2020-02-12:
I decided the inline asm version was just too ugly and unmaintainable, so I re-wrote it in C.  There's a single asm goto statement in the tx and rx code, but I think it it doesn't impact the ability of a C programmer to understand it.  I should have time to update calibration sketch in the next couple days.
https://github.com/nerdralph/picoUART

MCUdude, 2020-02-27:
@nerdralph sorry for the late reply, but I've been quite busy recently.

Your code looks very neat now! I just tried your test program, and it worked as expected.

This version does not have a function to deal with PROGMEM strings. Is the current implementation as efficient as it can get, or should your library handle these strings natively, like the previous version of picoUART did?

(Current implementation, master branch)
```c++
void HalfDuplexSerial::print(const __FlashStringHelper *ifsh)
{
  PGM_P p = reinterpret_cast<PGM_P>(ifsh);
  uint8_t c;

  do
  {
    c = pgm_read_byte(p++);
    TxByte(c);
  } while(c);
}
```

And it would also be great if you could implement an equivalent of the read_str() function, found in the [ReadASCIIString example](https://github.com/MCUdude/MicroCore/blob/master/avr/libraries/Serial_exampes/examples/ReadASCIIString/ReadASCIIString.ino).

nerdralph, 2020-02-27:
> @nerdralph sorry for the late reply, but I've been quite busy recently.
>
> Your code looks very neat now! I just tried your test program, and it worked as expected.
>
> This version does not have a [...]

That print method doesn't look right; it will print out the null terminator.  It should be like this:
while ( c = pgm_read_byte(p++) ) TxByte(c);

It could be slightly optimized, but only by one or two instructions.  I didn't like the idea having multiple copies of a prints_P, one for every UART version, so I left it out.  It's something I may come back to later if I come up with a generic way of doing it.

ReadASCIIString may have to wait until I finish working on a couple other little projects.

gamerpaddy, 2020-05-13:
Could a parity be added to this?  
I need, and theres unfourtainly no way around it, a Even parity bit in my 2400 baud serial tx.  (2400, SERIAL_8E1).
im feeding my attiny13a with an external 1mhz clock.
thanks!

nerdralph, 2020-05-14:
> 
> 
> Could a parity be added to this?
> I need, and theres unfourtainly no way around it, a Even parity bit in my 2400 baud serial tx. (2400, SERIAL_8E1).
> im feeding my attiny13a with an external 1mhz clock.
> thanks!

MicroCore uses my picoUART library, which does not support parity (only 8N1).  I'll consider adding it, and will include any updates in the following issue.
https://github.com/nerdralph/picoUART/issues/2

MCUdude, 2020-07-02:
I'm closing this issue since the current UART implementation now support single pin RX and TX.

closed 2020-07-02 by MCUdude