MCUdude/MicroCore #102: analogWrite bug?

nerdralph, 2020-03-19:
Shouldn't the pin be set to output even if SAFEMODE is off?

https://github.com/MCUdude/MicroCore/blob/afb826ff730a1a90c8ed70660aebc65b981836cb/avr/cores/microcore/wiring_pwm.c#L48

and shouldn't the mask be 1, not 2?

MCUdude, 2020-05-01:
actually, it's supposed to be 0x03, to mask out all the other bits. The T13 supports PWM on pin PB0 and PB1.

nerdralph, 2020-05-02:
> actually, it's supposed to be 0x03, to mask out all the other bits. The T13 supports PWM on pin PB0 and PB1.

But you are doing the AND before the _BV shift.  So the valid values of pin are 0 and 1, hence the code should be
> DDRB |= _BV(pin & 0x01)

I'd also check to make sure the compiler optimizes away the & operation when the pin is a constant, so it gets compiled to a single sbi instruction.

MCUdude, 2020-05-02:
Yes, of course. Brainfart from my side...

commit bc2055a1 (2020-05-02): Correct bit mask
Closes #102
closed 2020-05-02 by MCUdude