MCUdude/MicroCore #103: my Attiny13A require SCK pin (PB2) to pull to LOW to work properly.... I don't know what is going on...

GluTbl, 2020-04-16:
I program a simple blink sketch to blink... the sketch dont work... but work when i connect the SCK (PB2) to ground..... why i require this????.....

nerdralph, 2020-04-18:
My guess would be a bad ground connection.  Grounding PB2 would give a MCU ground of about 0.6V from the Vf of the input clamping diodes.
http://nerdralph.blogspot.com/2015/09/diodes-diodes-everywhere.html

GluTbl, 2020-05-01:
I don't know but some attiny13a work without putting the SCK to low... i think that one is faulty one.....

closed 2020-05-01 by GluTbl

MCUdude, 2020-05-01:
I recently tried to use an ATtiny13 that turned out to be faulty. On my chip. PB2 stayed low no matter what I did. I was trying to use it as a digital input. Swapping the chip for a different one solved the issue for me.

The application was a simple 230V switch with zero cross detection.