MCUdude/MicroCore #104: Q: ATtiny13A, Vbg as input?

rayholland, 2020-04-16:
Hi,

I power my ATtiny13A with an 18650 battery and would like to monitor battery voltage.
Do you know if i can use  Vbg as input with my ATtiny13A?

If so, could you please help me with a code snippet?

Thanks,

Best regards,
Raymond

nerdralph, 2020-04-18:
No, look at s. 14.12 of the datasheet.  There are only 4 ADC Mux inputs.
You might be able to get a crude measurement with the voltage drop over the internal diode on the RESET pullup.  Add a ~100k pulldown for a ~25uA current, which will give you a drop of ~450mV, plus ~45K * .025uA =~ 1125mV drop over the internal pullup.  When Vcc drops by 1V (say from 4V to 3V), the drop over the pullup will be proportionally more than the drop over the internal diode.  The change in Vf for diode is logarithmic with current, but for a resistor it is linear.

If you want a more accurate measurement, you could dedicate one of the ADC pins to a resistor divider, and use the internal BG reference (~1.115V).  If you use a ~1/4 divider (i.e. 150K & 470K), your battery voltage will be 4x your ADC reading.

p.s. when doing calculations with ADC values, avoid any floating point math as it will consume lots of flash.  Stick to uint16, and keep any divides or multiplies to powers of two so they can be done with simple bit shifts.

rayholland, 2020-04-18:
Thank you Ralph for your detailed info.

rayholland, 2020-04-18:
Question/issue can be closed.

closed 2020-04-18 by rayholland