MCUdude/MicroCore #105: (Serial.read blocking) error

DominicanMS, 2020-05-18:
Hello, try to compile the sketch "Menu Demo" and it presents the following error

`class HalfDuplexSerial has no member named read_blocking; did you mean read_char_blocking?`

MCUdude, 2020-05-19:
Really? I have no problems compiling this. Are you using a manual or board manager install?

`read_blocking` is defined here:
https://github.com/MCUdude/MicroCore/blob/817fd212431e01ade86be4f7290cd9a98fe69eb3/avr/cores/microcore/HalfDuplexSerial.h#L174

https://github.com/MCUdude/MicroCore/blob/817fd212431e01ade86be4f7290cd9a98fe69eb3/avr/cores/microcore/HalfDuplexSerial.cpp#L61-L69

DominicanMS, 2020-05-19:
I am using a manual install there is this is the error. Manager Manager is compiled successfully

MCUdude, 2020-05-19:
Then you'll have to update your manual installation. If you're using git, you can just pull it again. If not, re-download the zip-file

DominicanMS, 2020-05-19:
download the current version to confirm, throw the same compilation error, drop arduino, reinstall it again it does the same

MCUdude, 2020-05-19:
Then you would have to figure out where Arduino IDE finds these files. The latest boards manager release is even with the master branch, so a manual install is exactly the same as a boards manager install. Turn on verbose compilation in the IDE settings, trace down the HalfDuplexSerial.h+cpp file and check if read_blocking is present.

DominicanMS, 2020-05-20:
Thanks, it was a problem with temporary files in `C:\Users\PC\AppData\Local\arduino15`
delete it, configure it again and it worked correctly

closed 2020-05-20 by MCUdude