MCUdude/MicroCore #106: suggection and question about platformIO.md

Boteium, 2020-06-11:
In the platformIO template, the cpu clock is 96MHz rather than the 9.6Mhz it should be. 
`board_build.f_cpu = 96000000L`

This also doesn't work on my machine. I have to change it to "attiny13a" (all lower case)
`board = ATtiny13`

It seems that hfuse can be calculated automatically, but not lfuse. I have to put board_fuses.lfuse in my platformio.ini. Is this the expected result ?

Currently, I need to install the git version of the platformIO for MicroCore support. It would be nice to document this as well.
`platformio platform install git+https://github.com/platformio/platform-atmelavr.git`

MCUdude, 2020-06-11:
> In the platformIO template, the cpu clock is 96MHz rather than the 9.6Mhz it should be.
board_build.f_cpu = 96000000L

Thanks, that's clearly a typo.

> This also doesn't work on my machine. I have to change it to "attiny13a" (all lower case)
board = ATtiny13
It seems that hfuse can be calculated automatically, but not lfuse. I have to put board_fuses.lfuse in my platformio.ini. Is this the expected result ?

I think you have an older version of the atmelavr package installed. I just deleted the current microcore install and the atmelavr install, and everything works like it should. I can also calculate the fuses with the fuse script. However, I did fix a low fuse bug in the script a while ago.

> Currently, I need to install the git version of the platformIO for MicroCore support. It would be nice to document this as well.
platformio platform install git+https://github.com/platformio/platform-atmelavr.git

Nope, I'm using `platform = atmelavr`, and you shouldn't need git for this.

Boteium, 2020-06-12:
Is there a way to check if I have the correct version ?

I've uninstalled atmelavr and re-installed it again. l can confirm that lfuses can be calculated automatically now. 

But I still have problem setting the board name. I'm still using "attiny13a". Other non-working setting I've tried are "ATtiny13", "ATtiny13A", and "ATtiny13a".

MCUdude, 2020-06-12:
> Is there a way to check if I have the correct version ?

You can try `pio platform list`. I'm using atmelavr 2.1.0

Try to delete the PlatformIO MicroCore install too. Can you also post your the error message?

Boteium, 2020-06-12:
It says :

`Processing MicroCore (platform: atmelavr; framework: arduino; board: ATtiny13a)`
`Error: Unknown board ID 'ATtiny13a'`

pio platfor list says atmelavr Version: 2.2.0

After I uninstall the package. I did verify that ~/.platformio/packages/framework-arduino-avr-microcore does not exist.

If I use "attiny13a" rather than "ATtiny13a" after re-install atmelavr. It will download and install microcore v1.0.6.

`Processing MicroCore (platform: atmelavr; framework: arduino; board: attiny13)`
`----------------------------------------------------------------------------------------`
`PackageManager: Installing framework-arduino-avr-microcore @ ~1.0.6`
`framework-arduino-avr-microcore @ 1.0.6 has been successfully installed!`
`Verbose mode can be enabled via `-v, --verbose` option`
`CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny13.html`

MCUdude, 2020-06-12:
I'm not sure why `ATtiny13` doesn't work for you. @valeros probably know though

maujabur, 2020-07-10:
Hi!

I´m using platformio on windows, with Visual Studio code.
Atmel AVR is version 2.2.0

on my platformio.ini I use

```
[env:attiny13a]
platform = atmelavr
framework = arduino

board = attiny13a
```

When I hit built, **the fuses are not calculated correctly**, and always return as:

`avrdude: safemode: Fuses OK (E:FF, H:FF, L:6A)`

more details from the terminal:

```
PLATFORM: Atmel AVR 2.2.0 > ATtiny13A
HARDWARE: ATTINY13A 9MHz, 64B RAM, 1KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr-microcore 1.0.6
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 2.0
Building in release mode
Checking size .pio\build\attiny13a\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.3% (used 13 bytes from 64 bytes)
Flash: [========  ]  84.0% (used 860 bytes from 1024 bytes)
Configuring upload protocol...
AVAILABLE: stk500v1
CURRENT: upload_protocol = stk500v1
Looking for upload port...
Use manually specified: COM48
Uploading .pio\build\attiny13a\firmware.hex
```

MCUdude, 2020-09-08:
@Boteium if this is bothering you, I suggest you open an issue over at the PlatformIO AVR repo. It is after all the PlatfromIO implementation you're struggling with.

@maujabur the fuses won't calculate themselves. You'll have to run a command to calculate and set them:
```
pio run -t fuses
```
You'll also have to provide a set of parameters to get the fuse settings you want. This is documented in PlatformIO.md

closed 2020-09-08 by MCUdude

maujabur, 2020-09-29:
Maybe I didn´t express mysel correctly, or maybe it´s not a priority

But i´m under the impression that we have to declare several options under the platformio.ini that inform settings that affect fuses.

On the arduino IDE the fuses "calculate themselves", or more acurately the IDE calculate them, as it´s impossible for the fuses to take any action themselves (this expression is borderline rude)

So it was reasonable to me that some automation would calculate the fuses and set them on upload.

Anyway, I should have read the freaking documentation, even if for platform.io it´s confusing and incomplete.

But I thank you very much for your hard work, the code is mostly very good and you are doing it on your own time. I´ll consider making a donation if there is a way to do it.

Anyway, I´ll just use arduino IDE, as it´s more automated, even if the IDE has less resources.

I hope you have a nice day.

maujabur, 2020-09-29:
@MCUdude 
For me, buying one of the boards would be great, but it is prohibitive. Lots of import taxes and expensive shipping
I live in Brazil, and due to the situation the world is in, i´m not about to travel abroad.

I do have a collection of obards and used to buy them all the time, but now it´s not practical. also, my currency is now almost 6 to 1 relative to the dolar, and 6 months ago it was 4 to 1.

Also, I tried buying, but got the following: 

[image]

maujabur, 2020-09-29:
Actually, I bought this one a while ago! 

DIP-40 Arduino compatible development board