MCUdude/MicroCore #107: map function not working as expected

momonator25, 2020-06-20:
I tried to use a tiny13 to control a pwm output with the reading of the adc. For that i used the arduino map function to map the 10bit adc to the 8bit PWM timer. After no sucsess i hooked up my socpe and realised that the map function is generating not the expected output. I verified this by adding serial output to my test setup. It seems that the output of the map function is limited to 6bit and is overflowing. 

I have added the implementation of the "original" map function from the arduino reference to my code and compared the results.

```cpp
void setup() {
  OSCCAL = 79;
  Serial.begin(115200);
  Serial.println("Start");
  for (int i=0; i<=1024; i++){
    Serial.print("i: ");
    Serial.print(i);
  
    Serial.print(" incl. ");
    Serial.print(map(i, 0, 1023, 0, 255));
  
    Serial.print(" own ");
    Serial.println(my_map(i, 0, 1023, 0, 255));
    delay(1);
  }
}

long my_map(long x, long in_min, long in_max, long out_min, long out_max) {
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

void loop() { 
  while(true){
  }
}

```

Here the important part of serial output the described error.

```
....
i: 125 incl. 31 own 31
i: 126 incl. 31 own 31
i: 127 incl. 31 own 31
i: 128 incl. 31 own 31
i: 129 incl. -31 own 32
i: 130 incl. -31 own 32
....

```

MCUdude, 2020-06-21:
Thank you for reporting. That's certainly interesting... The current implementation is template based so that the compiler can pick the most suitable. Try this:

```c++
template <typename A, typename B, typename C, typename D, typename E> 
  int32_t newMap(A x, B in_min, C in_max, D out_min, E out_max)
  {
    return (int32_t)(x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
  }
```

All I've done is add a cast in the return statement.

momonator25, 2020-06-21:
> Thank you for reporting. That's certainly interesting... The current implementation is template based so that the compiler can pick the most suitable. Try this:
>
> ```c++
> template <typename A, [...]

Thanks for your reply. I will try this!

MCUdude, 2020-07-01:
Did it work for you?

momonator25, 2020-07-02:
> 
> 
> Did it work for you?

Sorry for the late reply. I just tried your suggested solution and it seems to work at least for my scenario. So I think the cast will do the trick and prevent from overflowing.

Thanks for your work and your very good implemenation!

closed 2020-07-02 by momonator25