MCUdude/MicroCore #109: Ultrasonic sensor issue

SaiSandeepKantareddy, 2020-07-07:
exit status 1
too few arguments to function 'uint32_t pulseIn(uint8_t, uint8_t, uint32_t)'
t
his is the issue for the below program using attiny13a

// defines pins numbers
const int trigPin = 2;
const int echoPin = 1;
const int buzzer = 3;
const int ledPin = 0;

// defines variables
long duration;
int distance;
int safetyDistance;

void setup() {
pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
pinMode(echoPin, INPUT); // Sets the echoPin as an Input
pinMode(buzzer, OUTPUT);
pinMode(ledPin, OUTPUT);
Serial.begin(9600); // Starts the serial communication
}

void loop() {
// Clears the trigPin
digitalWrite(trigPin, LOW);
delayMicroseconds(2);

// Sets the trigPin on HIGH state for 10 micro seconds
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);

// Reads the echoPin, returns the sound wave travel time in microseconds
duration = pulseIn(echoPin, HIGH);

// Calculating the distance
distance= duration*0.034/2;

safetyDistance = distance;
if (safetyDistance <= 5){
  digitalWrite(buzzer, HIGH);
  digitalWrite(ledPin, HIGH);
}
else{
  digitalWrite(buzzer, LOW);
  digitalWrite(ledPin, LOW);
}

// Prints the distance on the Serial Monitor
Serial.print("Distance: ");
Serial.println(distance);
}

maujabur, 2020-07-10:
try

`duration = pulseIn(echoPin, HIGH,1000);`

SaiSandeepKantareddy, 2020-07-12:
> try
> 
> `duration = pulseIn(echoPin, HIGH,1000);`

Hi, it says out of memory on atiny13a.

maujabur, 2020-07-13:
So it must be too big for the flash on the attiny13. Try removing some functionality. Serial usually eats a lot of memory, for example.
If you are using the arduino IDE, on the black region below the code it usally roports the size of the program after compiled.

Maybe it´s best if you switch to attiny85, it has "lots" of flash

SaiSandeepKantareddy, 2020-07-15:
> So it must be too big for the flash on the attiny13. Try removing some functionality. Serial usually eats a lot of memory, for example.
> If you are using the arduino IDE, on the black region below [...]

Thanks for that, will work on it and update you.

mcuee, 2023-05-12:
Old issue like this can probably be closed.

closed 2023-05-12 by MCUdude