MCUdude/MicroCore #110: ADC1

adamalfath, 2020-07-10:
Hi,
I'm having trouble reading value from ADC1 (PB2). I write this code `analogWrite(0,analogRead(2)/4);` to 'echo' the reading to PWM output driving LED brightness since I'm not activated the UART.
I'm reading ACS712 sensor and it's confirmed having 2.5V on PB2 using multimeter, however I didn't observe any change on the LED. I'm also have trimpot input at ADC3 (PB3) and the 'echo' just working fine, the brightness change according to trimpot input.

adamalfath, 2020-07-10:
Oh wait, so analogRead suppose to be calling the ADC channel number not the port number?, but other operation (digitalRead, digitalWrite, analogWrite) is fine to use port number? 
https://github.com/MCUdude/MicroCore/blob/9199e45eb749d34c4f73b5112b62bad5698a1906/avr/variants/attiny13/pins_arduino.h#L71
That trimpot input just happy coincidence because port number is same as ADC channel number.
If so, I just need confirmation then :grin:

nerdralph, 2020-07-10:
Yup, the analog pin numbering is not very intuitive.  On the t13 (and t85), PB0 & PB1 are not ADC inputs, only PB2, PB3, PB4, & PB5.

closed 2020-07-11 by adamalfath