MCUdude/MicroCore #111: Wiring issue in "MicroCore ATtiny13 minimal setup"?

wmzart, 2020-08-02:
In the MicroCore ATtiny13 minimal setup, shown here:

  https://github.com/MCUdude/MicroCore#minimal-setup

the ISP PROGRAMMING HEADER section shows at J1, pin 3: MOSI and at pin 4: SCK.

On my programmer however, Avr Dragon, pin 3 is SCK and pin 4 is MOSI, shown in section "3.3.1.1. SPI (ISP) Header (mounted)":

[image]

(Source: http://ww1.microchip.com/downloads/en/devicedoc/atmel-42723-avr-dragon_userguide.pdf )

Are pins 3 and 4 wrongly swapped on the schematic drawing?

MCUdude, 2020-08-02:
Thanks for reporting, you're absolutely right. Luckily, the ATtiny development board I sell does have the ISP wiring correct 🙂.
I'll fix it right away!

wmzart, 2020-08-02:
I am glad the development board is correct. Strange though that no one noticed this before.

commit 8da3c722 (2020-08-02): Fix incorrect ISP connector wiring

For the minimal setup wiring diagram. Closes #111
closed 2020-08-02 by MCUdude