MCUdude/MicroCore #120: How to update MicroCore last version in platformio ?

Witawat, 2020-12-20:
How to update MicroCore last version in platformio ?

platformio use version 1.0.6 but i found in github version 2.0.2
i need update in platformio new version

thankyou

MCUdude, 2020-12-20:
Hi!

It's correct that PlatformIO hasn't been updated yet. I will notify the developers, but you should know that 1.0.6 is very stable and proven. I did introduce some breaking changes in 2.0.0, but most of these were removed in 2.0.2 because they caused a lot of trouble.

While you wait for 2.0.2,  you can safely use 1.0.6. It's just as good 🙂

Witawat, 2021-01-04:
> Hi!
>
> It's correct that PlatformIO hasn't been updated yet. I will notify the developers, but you should know that 1.0.6 is very stable and proven. I did introduce some breaking changes in 2.0.0, but [...]

now i update to 2.0.2 .. 
in version have warning

int Ldr_pin = PB3;

 warning: invalid conversion from 'int' to 'analog_pin_t' [-fpermissive]
   mapLDR = map(analogRead(Ldr_pin), 0, 1023, 0, 10);

this bug ?
                                  ^

MCUdude, 2021-01-04:
You can't refer to analog pins with PB0..PB5. You have to use A0, A1, A2, or A3. That's because there is a mismatch between the analog and digital pin number.

MCUdude, 2021-01-04:
use 

``` c

analog_pin_t Ldr_pin = A3;

mapLDR = map(analogRead(Ldr_pin), 0, 1023, 0, 10);

```

closed 2021-01-04 by MCUdude