MCUdude/MicroCore #122: analogRead compile errors

pokearoo, 2021-01-13:
The following program works well with version 1.0.7 of MicroCore. It's burnt in an IC and runs well. However I updated to version 2.0.2 and I get compile errors. In fact any versions more recent than 1.0.7 produce the same error.

"invalid conversion from 'int' to 'analog_pin_t' [-f permissive]

Thanks in advance
Luc

int Ch1=1;  //Digital output 1
int Ch2=0;  //Digital output 2
int Pot1=A2; //Analogue input 1
int Pot2=A3; //Analogue input 2
int Val1=0; //Analogue reading 1
int Val2=0; //Analogue reading 2

void setup() {
  // put your setup code here, to run once:

pinMode(Ch1,OUTPUT);
pinMode(Ch2,OUTPUT);

}

void loop() {
Val1 = analogRead(Pot1);
delay(2);
if (Val1<5){
  Val1=0;
}
analogWrite(Ch1,Val1/4);  //PWM output 0 to 255

Val2 = analogRead(Pot2);
  if (Val2<5){
  Val2=0;
  }
delay(2);
analogWrite(Ch2,Val2/4); //PWM output 0 to 255

MCUdude, 2021-01-13:
MicroCore v2.0.0 and newer introduced a breaking change when handling analog pins.

use `analog_pin_t` instead of int for assigning analog pins:

```c
int Ch1=1; //Digital output 1
int Ch2=0; //Digital output 2
analog_pin_t Pot1=A2; //Analogue input 1
analog_pin_t Pot2=A3; //Analogue input 2
int Val1=0; //Analogue reading 1
int Val2=0; //Analogue reading 2
```

closed 2021-01-13 by MCUdude

pokearoo, 2021-01-13:
That works.

Thanks

AchimPieters, 2021-01-17:
@MCUdude I got the same error with this code:

```
//#define testrun // uncomment for test run the motor

#define LEFTSENSOR     3 // PIN 2
#define LEFTSENSORADC  3 // also PIN 2
#define LEFTMOTOR      4 // PIN 3
//#define LEFTDUTY     219 // for calibrate motor in same speed
#define RIGHTSENSOR    2 // PIN 7
#define RIGHTSENSORADC 1 // also PIN 7
#define RIGHTMOTOR     1 // PIN 6
//#define RIGHTDUTY    195 // for calibrate motor in same speed

#define THRESHOLD     56 // readings threshold for line detection
// Every analog readings from 0-1023,
// so an unsigned int allow accumulate 64 reads for average.
// Setting THRESHOLD 56 means white average * 56 / 64
// i.e. readings below 7/8 of white average means a line

unsigned int leftThreshold;
unsigned int rightThreshold;

unsigned int leftRead;
unsigned int rightRead;

bool lastTurnLeft = true;
bool firstMeetBlack = true;

void setup() {
  pinMode(LEFTMOTOR, OUTPUT);
  pinMode(RIGHTMOTOR, OUTPUT);

  delay(2000);

  readSensor(THRESHOLD);
  leftThreshold = leftRead >> 6; // divide by 64
  rightThreshold = rightRead >> 6; // divide by 64
}

void loop() {
#ifdef testrun
  turn(true, 0, 10); // minor turn left
  delay(500);
  turn(false, 0, 10); // minor turn right
  delay(500);
  turn(true, 0, 50); // turn left
  delay(500);
  turn(false, 0, 50); // turn right
  delay(500);
  turn(true, 0, 255); // full turn left
  delay(500);
  turn(false, 0, 255); // full turn right
  delay(500);
  turn(true, 10, 40); // minor dash and turn
  delay(500);
  turn(true, 30, 120); // dash and turn
  delay(500);
  turn(true, 63, 255); // full dash and turn
  delay(500);
  turn(true, 10, 0); // minor dash
  delay(500);
  turn(true, 50, 0); // dash
  delay(500);
  turn(true, 255, 0); // full dash
  delay(3000);
#else // normal run
// ========== real code start here ==========
  readSensor(4);
  leftRead >>= 2; // divide by 4
  rightRead >>= 2; // divide by 4

  bool leftBlack = (leftRead < leftThreshold);
  bool rightBlack = (rightRead < rightThreshold);

  if (firstMeetBlack && (leftBlack || rightBlack)) {
    // try to land on the line when first found it
    turn(false, 40, 160); // dash and turn
    firstMeetBlack = false;
    // todo: try to detect lost the line and set it to true again
  } else {
    if (leftBlack && rightBlack) {
      turn(lastTurnLeft, 0, 40);
    } else if (leftBlack) {
      turn(true, 0, 40);
    } else if (rightBlack) {
      turn(false, 0, 40);
    } else {
      turn(leftRead & 0b1, 0, 40); //use left readings as random run
    }
  }
  delay(40);
#endif //normal run
}

void turn(bool turnLeft, byte initTime, byte turnTime) {
  lastTurnLeft = turnLeft;

  motor(true, true);
  delay(initTime);

  motor(!turnLeft, turnLeft);
  delay(turnTime);

  motor(false, false);
}

void motor(bool leftOn, bool rightOn) {
#ifdef LEFTDUTY // PWM
  analogWrite(LEFTMOTOR, leftOn?LEFTDUTY:0);
#else
  digitalWrite(LEFTMOTOR, leftOn?HIGH:LOW);
#endif

#ifdef RIGHTDUTY // PWM
  analogWrite(RIGHTMOTOR, rightOn?RIGHTDUTY:0);
#else
  digitalWrite(RIGHTMOTOR, rightOn?HIGH:LOW);
#endif
}

void readSensor(byte count) {
  leftRead = 0;
  rightRead = 0;
  for (int i = 0; i < count; i++) {
    leftRead += analogRead(LEFTSENSORADC);
    rightRead += analogRead(RIGHTSENSORADC);
    delay(1);
  }
}
```

MCUdude, 2021-01-17:
@AchimPieters the same solution applies for your code as well. Use analog_pin_t.