MCUdude/MicroCore #124: Invalid device signature for Attiny 13a

varadhas, 2021-03-03:
I am getting the following error while  uploading a sketch:

       avrdude: Yikes!  Invalid device signature.
       Double check connections and try again, or use -F to override
       this check.

MCUdude, 2021-03-03:
Turn on verbose upload in the IDE settings and post the output. If you're getting a device signature of 0x000000 or 0xffffff, you're probably wired it incorrectly.

varadhas, 2021-03-03:
Thanks for ur valuable time;

Arduino: 1.8.13 (Windows 10), Board: "ATtiny13, 1.2 MHz internal osc., EEPROM retained, BOD 2.7V, Micros enabled"

Sketch uses 188 bytes (18%) of program storage space. Maximum is 1024 bytes.

Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for local variables. Maximum is 64 bytes.

C:\Users\VARADHA\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf -v -pattiny13 -cstk500v1 -PCOM4 -b19200 -Uhfuse:w:0xfb:m -Ulfuse:w:0b00101010:m -Uflash:w:C:\Users\VARADHA\AppData\Local\Temp\arduino_build_609256/Blink.ino.hex:i 

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf"

         Using Port                    : COM4

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATtiny13

         Chip Erase delay              : 4000 us

         PAGEL                         : P00

         BS2                           : P00

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff

           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

An error occurred while uploading the sketch

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000

avrdude: Yikes!  Invalid device signature.

         Double check connections and try again, or use -F to override

         this check.

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

varadhas, 2021-03-03:
Arduino: 1.8.13 (Windows 10), Board: "ATtiny13, 1.2 MHz internal osc.,
EEPROM retained, BOD 2.7V, Micros enabled"

Sketch uses 188 bytes (18%) of program storage space. Maximum is 1024 bytes.

Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for
local variables. Maximum is 64 bytes.

C:\Users\VARADHA\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude
-CC:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf
-v -pattiny13 -cstk500v1 -PCOM4 -b19200 -Uhfuse:w:0xfb:m
-Ulfuse:w:0b00101010:m
-Uflash:w:C:\Users\VARADHA\AppData\Local\Temp\arduino_build_609256/Blink.ino.hex:i

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is
"C:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf"

         Using Port                    : COM4

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATtiny13

         Chip Erase delay              : 4000 us

         PAGEL                         : P00

         BS2                           : P00

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page
          Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW
 MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------
----- ----- ---------

           eeprom        65     5     4    0 no         64    4      0
 4000  4000 0xff 0xff

           flash         65     6    32    0 yes      1024   32     32
 4500  4500 0xff 0xff

           signature      0     0     0    0 no          3    0      0
0     0 0x00 0x00

           lock           0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00

           calibration    0     0     0    0 no          2    0      0
0     0 0x00 0x00

           lfuse          0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

An error occurred while uploading the sketch

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000

avrdude: Yikes!  Invalid device signature.

         Double check connections and try again, or use -F to override

         this check.

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MCUdude, 2021-03-03:
> avrdude: Device signature = 0x000000

You're having a hardware issue, the programmer is not able to communicate with the chip for some reason. I suggest you go over your wiring and ask on the Arduino forum if you need further assistance.

varadhas, 2021-03-03:
thank u...

varadhas, 2021-03-03:
Actually it was worked . after i am trying to upload a different sketck it started showing above stated error.

varadhas, 2021-03-03:
Actually it was worked . after I am trying to upload a different sketck  using utrasonic sensor, I changed some settings like the clock to 9.6 Mhz  and enabled micros. after that it started showing above stated error. But my last updated blink program is working still now. if i try to upload any new project, even blink project its not working

nerdralph, 2021-03-03:
> 
> 
> Actually MicroController was worked for Blink Project . After I am trying to upload a different sketck for working with utrasonic sensor, I changed some settings like the clock to 9.6 Mhz and enabled micros. after that it started showing above stated error. But my last updated blink Project is working still now. if I try to upload any new project, even blink project its not getting uploaded

Which, as Hans said, is indicative of connection problems.  My first guess would be a loose connection to the t13 RST pin.  For debugging you should run avrdude from the command line.

varadhas, 2021-03-03:
Actually I don't know  how to debug in command line , but I executed the project in command line .  I got same error message , shown below:

C:\Users\VARADHA\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18\bin>avrdude -CC:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf -v -pattiny13 -cstk500v1 -PCOM3 -b19200 -Uhfuse:w:0xfb:m -Ulfuse:w:0b00101010:m -Uflash:w:C:\Users\VARADHA\AppData\Local\Temp\arduino_build_609256/Blink.ino.hex:i

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\VARADHA\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.0.3/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

C:\Users\VARADHA\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18\bin>

varadhas, 2021-03-03:
[image]
Circuit diagram image

WrichikBasu, 2021-03-27:
This should be posted in Arduino forum or some similar place, but since this issue hasn't yet been closed, let me add something. The error you are getting is definitely a wiring problem. For the past few months, I have been programming ATmega 8A uCs using minicore, and I was getting this same problem at the beginning. It turned out to be a wiring problem. Some tips from my end:
1. Use ICSP header pins on the programmer Arduino. These pins remain constant over all Arduinos (be it Nano, Uno or Mega), and hence can be used unambiguously. See [here](https://www.circuitstoday.com/wp-content/uploads/2018/02/Arduino-Nano-ICSP.jpg) for ICSP header pinout.
1. Put a 100 nF capacitor between VCC and GND of the ATiny.
1. The programmer Arduino should be in RESET stage when it is programming the ATtiny. Connect a 22 uF capacitor between the RESET and GND of the nano.
1. Make sure you choose "Arduino as ISP (Microcore)" under Programmer before uploading the program. Also, use "Upload using programmer" under Sketch menu in the Arduino IDE. The Nano should have the ArduinoISP sketch loaded into it.

nerdralph, 2021-03-28:
I've never seen a case where adding the .1uF cap has solved a problem for a 5V powered AVR.  While it's not a bad idea to use one, it is a bad idea to propagate the idea that the lack of a decoupling cap on a t13 (or similar tiny AVRs) is a likely source of problems.

WrichikBasu, 2021-03-28:
> it is a bad idea to propagate the myth that the lack of a decoupling cap on a t13 (or similar tiny AVRs) is a likely source of problems

That capacitor is definitely not a source of problems, but I just mentioned it here. Probably should have been more clear about it.

mcuee, 2022-12-25:
Since this is a hardware issue, it can be closed.

closed 2023-01-09 by MCUdude