MCUdude/MicroCore #126: MIDI 31250 Baud Rate

sharpie7, 2021-04-19:
Thanks for creating this package. It really helped me out on a project this weekend.

Quick question/issue: MIDI uses a fixed Baud rate of 31250 (specified in the standard). Is support for this rate possible using the serial features of this library? If so, could the compatibility with the different CPU clocks be added to the table in core_settings.h ?

MCUdude, 2021-04-19:
Hi! Great to hear you find MicroCore useful!

A baud rate of 31250 should absolutely work, even though I haven't tried it. If you run the ATtiny13 at 9.6 MHz you should be good. All you need to do is to change the baud rate number in core_settings.h. I'll see if I can test it myself and figure out which clock speeds support 31250 baud.

sharpie7, 2021-04-19:
Many thanks.

MCUdude, 2021-08-14:
31250 baud will work on all clock frequencies except for 600kHz and 128kHz.

closed 2021-08-14 by MCUdude