MCUdude/MicroCore #132: Weird operator compilation error

BlvckBytes, 2021-09-02:
I'm trying to compile the following code with your board enabled:

```
int* numbers;

void setup() {
 numbers = new int[3] {0, 1, 2};
}

void loop() {}
```

It compiles fine with a UNO or ESP board. With the ATTiny13 loaded, I get the following error:

```
Arduino: 1.8.13 (Mac OS X), Board: "ATtiny13, 9.6 MHz internal osc., EEPROM retained, BOD 2.7V, Micros disabled"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/blvckbytes/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/blvckbytes/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/blvckbytes/Documents/Arduino/libraries -fqbn=MicroCore:avr:13:clock=9M6,eeprom=keep,BOD=2v7,timing=micros_disabled -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172 -warnings=none -build-cache /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_cache_364622 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -verbose /Users/blvckbytes/Documents/Arduino/sketch_sep02a/sketch_sep02a.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/blvckbytes/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/blvckbytes/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/blvckbytes/Documents/Arduino/libraries -fqbn=MicroCore:avr:13:clock=9M6,eeprom=keep,BOD=2v7,timing=micros_disabled -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172 -warnings=none -build-cache /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_cache_364622 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -verbose /Users/blvckbytes/Documents/Arduino/sketch_sep02a/sketch_sep02a.ino
Using board '13' from platform in folder: /Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0
Using core 'microcore' from platform in folder: /Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0
Detecting libraries used...
/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10813 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/cores/microcore -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/variants/attiny13 /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch/sketch_sep02a.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10813 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/cores/microcore -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/variants/attiny13 /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch/sketch_sep02a.ino.cpp -o /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10813 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto -g -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/cores/microcore -I/Users/blvckbytes/Library/Arduino15/packages/MicroCore/hardware/avr/2.1.0/variants/attiny13 /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch/sketch_sep02a.ino.cpp -o /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch/sketch_sep02a.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_cache_364622/core/core_7ddd98c88bffa12159ca129d2d1a96cb.a
Linking everything together...
/Users/blvckbytes/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -w -Os -Wl,--gc-sections -mmcu=attiny13a -w -flto -g -o /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch_sep02a.ino.elf /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/sketch/sketch_sep02a.ino.cpp.o /var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172/../arduino_cache_364622/core/core_7ddd98c88bffa12159ca129d2d1a96cb.a -L/var/folders/jt/494334l53g177m6s43fh86_w0000gn/T/arduino_build_594172 -lm
/var/folders/jt/494334l53g177m6s43fh86_w0000gn/T//ccSPhbD0.ltrans0.ltrans.o: In function `setup':
/Users/blvckbytes/Documents/Arduino/sketch_sep02a/sketch_sep02a.ino:4: undefined reference to `operator new[](unsigned int)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny13.
```

Any ideas on why that is?

MCUdude, 2021-09-03:
To be honest, I have no idea why it doesn't work. I've not done anything to prevent it from working with MicroCore, and it does work with my other cores.

It looks almost like there is a built-in limitation for low memory devices like the ATtiny13.

But why use dynamic memory allocation on the ATtiny13 in the first? In general, it's bad practice to do this on small microcontrollers, and the ATtiny13 is definitely the smallest of the smallest!

BlvckBytes, 2021-09-03:
You are entirely right! :) I just needed a few hours of getting into C, and developing the proper mindset. I now know why dynamic mem alloc is nonsense, especially on this device.

I guess that this is because the C++ stdlib doesn't fit on a device this small, and thus the new operator, which is really malloc, is not defined. Pulling in malloc alone almost fills this thing up, lol. I had no idea how small 1KB really is.

Thanks, and sorry for my uninformed question, should've learned more before this, xD.

MCUdude, 2021-09-03:
> I had no idea how small 1KB really is.

1kB is the program memory, and with clever code, you can actually fit quite a lot. However, 64 _bytes_ of RAM get eaten away very quickly if you're not careful.

A rule of thumb I follow is to _never_ use dynamic memory allocation on microcontrollers if I absolutely don't have to. A microcontroller is only doing one thing, running its tiny program, and as long as there is RAM available it doesn't really matter if you create static arrays that are bigger than they really need to be, so static allocation is most of the time just fine.

Dynamic memory allocation makes very good sense on a computer, but not so much on a weak 8-bitter 🙂

closed 2021-09-03 by MCUdude