MCUdude/MicroCore #141: Automatically optimize away micros()

MCUdude, 2022-12-27:
@nerdralph did something clever in the current millis() implemenation. There's no need to manually enable millis in the tools menu or by passing a flag to the compiler, because this is done by avr-gcc at compile time.

It would be _really, really_ neat if there was a way to do the same for micros()! I'm not necessarily talking about converting it to assembly, but instead, having _all_ micros() related code go away if micros() isn't used in the user program.

As a start, I've moved the ISR and all micros() related code into a separate file, micros.cpp. Now all code within this file is automatically left out by the compiler if no micros() is present, but I'm really struggling to get rid of the micros initialization code.

Before micros() can run, timer0 needs to be initialized. At the moment this happens in init(). Sadly I haven't figured out how to make the compiler automatically pull in the timer0 initialization code whenever micros() is used in the user program.

@nerdralph how did you manage to do just this with your millis() implementation? And is there hope for my implementation of micros() without converting everything into assembly, which I'm not able to do?

MCUdude, 2022-12-27:
I've also tried to play around with `.initN`, but for some reason, it gets optimized away no matter what I do.

``` c
void init_micros(void) __attribute__ ((naked)) __attribute__ ((section (".init6")));

void micros()
{
    // Set a suited prescaler based on F_CPU
    #if F_CPU >= 4800000L
      TCCR0B = _BV(CS00) | _BV(CS01); // F_CPU/64
    #else
      TCCR0B = _BV(CS01);             // F_CPU/8
    #endif
    // Enable overflow interrupt on Timer0
    TIMSK0 = _BV(TOIE0);
    // Set timer0 couter to zero
    TCNT0 = 0;
}

nerdralph, 2022-12-28:
assembler isn't necessary to link in code only when used.  The key is more
modular code where everything isn't in one big .a with a lot of global
dependencies.  I talked about it a bit in my post on ArduinoShrink:
http://nerdralph.blogspot.com/2021/04/honey-i-shrunk-arduino-core.html

As I've mentioned before, feel free to copy code from picoCore.
https://github.com/nerdralph/picoCore
I'm hoping to get some time over the holidays to fix the asm constraints
bug with the delay() function that happens with more recent versions of
gcc, but otherwise I don't think I'll be doing any embedded software work.

MCUdude, 2023-01-03:
Thanks! I actaully figured out how I could have code that ran _before_ main() and still be optimized out if micros were not used in the user application.

I moved all micros() related functions into micros.cpp, including it's init function:

https://github.com/MCUdude/MicroCore/blob/5c402265000a3d834999324756df6d52a694a3ca/avr/cores/microcore/micros.cpp#L7-L21

> As I've mentioned before, feel free to copy code from picoCore.

Thank you! There's one thing that alloys me. When a blank sketch are compiled with picoCore, it compiles down to 26 bytes using avr-gcc 7.3.0. However, with MicroCore it's 42 bytes. I've tried to copy exactly the settings you have in your platform.txt file, and the two cores are using the exact same compiler. How is this even possible?

nerdralph, 2023-01-03:
Nice work with the new micros.cpp.  It may be worthwhile adding a comment explaining how the linker will only pull in the .o (including the init6 code) when micros() is an unresolved symbol somewhere else.  Even though I've used the same trick myself before, it took a minute of looking at it before I remembered how it works.

The reason for picoCore's smaller size is pcrt.S.  You included the code but didn't update platform.txt to force the compiler to use it instead of the default init code from avr-libc.
https://github.com/nerdralph/picoCore/blob/master/avr/platform.txt#L52

MCUdude, 2023-01-03:
The weird thing was that even though I had an exact 1:1 copy of your platform.txt file it wouldn't work. However, after looking closely at the Arduino ICE verbose compilation output, I realized that I had some old leftover LTO flags in boards.txt that was overriding the compiler.c.elf.extra_flags in platform.txt. Now I'm down to 28 bytes, which is much better. Thank you for your help!

https://github.com/MCUdude/MicroCore/blob/5c402265000a3d834999324756df6d52a694a3ca/avr/boards.txt#L31-L36

closed 2023-01-03 by MCUdude