MCUdude/MicroCore #143: Error compiled for attinu13A

EmSerg74, 2023-01-12:
Hi, my code:
`
int main(void){
uint8_t course;
uint8_t course_old;
PORTB = B0010000;
DDRB  = B00001111;
while(1){
course = (uint8_t) bitRead(PINB, 4);
if(course != course_old){
  delayMicroseconds(30);
  course_old = course;
}
if(course>0){
  
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
  PORTB = B00010001;
 
  PORTB = B00010000;
  
  PORTB = B00010010;
  PORTB = B00010010;
  PORTB = B00010010;
  PORTB = B00010010;
  PORTB = B00010010;
  PORTB = B00010010;
  PORTB = B00010010;
[... 821 lines cut ...]
C:\Users\USER\AppData\Local\Temp\arduino_modified_sketch_476804/sketch_jan12a.ino:203
  PORTB = B00011000;
 1b4:	88 bb       	out	0x18, r24	; 24
C:\Users\USER\AppData\Local\Temp\arduino_modified_sketch_476804/sketch_jan12a.ino:204
  PORTB = B00011000;
 1b6:	88 bb       	out	0x18, r24	; 24
C:\Users\USER\AppData\Local\Temp\arduino_modified_sketch_476804/sketch_jan12a.ino:205
  PORTB = B00011000;
 1b8:	88 bb       	out	0x18, r24	; 24
 1ba:	a1 cf       	rjmp	.-190    	; 0xfe <__stack+0x5f>

000001bc <__bad_interrupt>:
__vector_1():
 1bc:	21 cf       	rjmp	.-446    	; 0x0 <__vectors>

000001be <_exit>:
exit():
/home/jenkins-mingw32/workspace/avr-gcc-staging/label/Ubuntu14.04x64-mingw32/gcc-build/avr/avr25/tiny-stack/libgcc/../../../../../gcc/libgcc/config/avr/lib1funcs.S:2278
 1be:	f8 94       	cli

000001c0 <__stop_program>:
__stop_program():
/home/jenkins-mingw32/workspace/avr-gcc-staging/label/Ubuntu14.04x64-mingw32/gcc-build/avr/avr25/tiny-stack/libgcc/../../../../../gcc/libgcc/config/avr/lib1funcs.S:2280
 1c0:	ff cf       	rjmp	.-2      	; 0x1c0 <__stop_program>
`

MCUdude, 2023-01-12:
I don't know what you want me to do.

Try adding `cli();` before your loop, and replace `delayMicroseconds`, which is interrupt based, with `_delay_us`.

EmSerg74, 2023-01-15:
This foto is for the first post:
[image]

**If you use this code:**
`// PRG_2 (MCUDUDE MicroCore 2.2.0 (ATINY13); int9.6MHz; Micros enable)
// PB4=1; PB0=644kHz; 0=1.44us; 1=104ns.
// PB4=0; PB0=540kHz; 0=1.52us; 1=320ns.
int main(void){
uint8_t course;
uint8_t course_old;
PORTB = B00010000;
DDRB  = B00001111;
while(1){
course = (uint8_t) bitRead(PINB, 4);	// ?=if not (uint8_t) then 32bit!
if(course != course_old){
	delayMicroseconds(30);
	course_old = course;
}
if(course>0){  
  PORTB = B00010001;
  PORTB = B00010000;  
  PORTB = B00010010;
  PORTB = B00010000;
} else {
  PORTB = B00010100;
  PORTB = B00010000;
  PORTB = B00011000;
  PORTB = B00010000;
}
}
}`
**then the result will be like this:**
[image]
[image]

**If you delete delay:**

`// PRG_2 (MCUDUDE MicroCore 2.2.0 (ATINY13); int9.6MHz; Micros disable)
// PB4=1; PB0=1207kHz; 0=720ns;  1=104ns.
// PB4=0; PB0= 880kHz; 0=816ns;  1=312ns.
int main(void){
uint8_t course;
uint8_t course_old;
PORTB = B00010000;
DDRB  = B00001111;
while(1){
course = (uint8_t) bitRead(PINB, 4);	// ?=if not (uint8_t) then 32bit!
if(course != course_old){
//	delayMicroseconds(30);                 //  
	course_old = course;
}
if(course>0){  
  PORTB = B00010001;
  PORTB = B00010000;  
  PORTB = B00010010;
  PORTB = B00010000;
} else {
  PORTB = B00010100;
  PORTB = B00010000;
  PORTB = B00011000;
  PORTB = B00010000;
}
} 
}`

**then we will get this result:**
[image]
[image]

**I understand that the ARDUINO IDE code is much larger and contains a lot of garbage, so it is very slow. Why use conversion if it works chaotically? People think that their code works correctly, but they are mistaken and spend time on it, and then they get disappointed. Why give hope in this way that they will be able to learn programming?**

EmSerg74, 2023-01-15:
Even here, bug. 
I used the code insertion above, but the page using the photo ignores the code insertion and removes the line feed in the code.

EmSerg74, 2023-01-15:
I think it's understandable. When PB4 = 0 or 1 on the pin, the signal on PB0-PB3 should have the same frequency and waveform - almost a meander due to the setting frequency of 9.6MHz. But this is not happening. 
Maybe I should use the WinAVR program instead of the cli()?

MCUdude, 2023-01-15:
> I understand that the ARDUINO IDE code is much larger and contains a lot of garbage, so it is very slow. Why use conversion if it works chaotically? People think that their code works correctly, but they are mistaken and spend time on it, and then they get disappointed.

That's why 3rd party cores exist. MicroCore is pretty much as efficient as C code can get for AVR. I also tend not to discuss with people that are not interested in a solution, and instead just want to be rude.

if you don't use `void setup()` and `void loop()` in your code, but instead use `int main()`, the compiler won't include _any_ Arduino code, and you're working bare metal. What you decide to add to your program is your choice. You won't get significantly better performance by using WinAVR or Atmel Studio.

The only "Arduino code" in this program is bitRead, which is just this macro: `#define bitRead(value, bit)  (((value) >> (bit)) & 0x01)`. If your code is slow, you're to blame.

delayMicroseconds is a timer-based function, and there are good reasons to sometimes use this instead of _delay_us.

```c
int main(void){
uint8_t course;
uint8_t course_old;
PORTB = B00010000;
DDRB  = B00001111;
while(1){
course = (uint8_t) bitRead(PINB, 4);	// ?=if not (uint8_t) then 32bit!
if(course != course_old){
//	delayMicroseconds(30);                 //  
	course_old = course;
}
if(course>0){  
  PORTB = B00010001;
  PORTB = B00010000;  
  PORTB = B00010010;
  PORTB = B00010000;
} else {
  PORTB = B00010100;
  PORTB = B00010000;
  PORTB = B00011000;
  PORTB = B00010000;
}
} 
}
```

If you really want code that can toggle a pin really fast, try this instead:

```c
int main()
{
  DDRB |= 1 << PB2;
  while(1)
  {
    PINB = 1 <<PB2;
  }
}
```

EmSerg74, 2023-01-15:
I know about such a fast code, but it doesn't fit me by definition. I need to generate a signal with a frequency of 100kHz on two pins. When switching the input signal level, the signal generation should be on the other two pins. When switching the input signal, there should be a delay and on all pins the signal = 0. The maximum frequency on the Arduino IDE is 79kHz and 82kHz, depending on the input signal. It is the difference in the generated frequencies that creates the problem. With frequent switching, the difference of 3kHz is an audible sound range. 
This is the main problem, and the second problem is the inability to reach a frequency of 100 kHz using an internal quartz at 9.6 MHz.

EmSerg74, 2023-01-15:
for information:
2021 Microchip Technology Inc. Data Sheet Complete DS40002307A-page 110 
https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny13A-Data-Sheet-DS40002307A.pdf

WinAVR-20100110\avr\include\avr\iotn2313a.h 
/* Low Fuse Byte */
#define FUSE_CKSEL0 (unsigned char)~_BV(0)  /* Select Clock Source */
#define FUSE_CKSEL1 (unsigned char)~_BV(1)  /* Select Clock Source */
#define FUSE_SUT0 (unsigned char)~_BV(2)  /* Select start-up time */
#define FUSE_SUT1 (unsigned char)~_BV(3)  /* Select start-up time */
#define FUSE_CKDIV8 (unsigned char)~_BV(4)  /* Start up with system clock divided by 8 */
#define FUSE_WDTON (unsigned char)~_BV(5)  /* Watch dog timer always on */
#define FUSE_EESAVE (unsigned char)~_BV(6)  /* Keep EEprom contents during chip erase */
#define FUSE_SPIEN (unsigned char)~_BV(7)  /* SPI programming enable */
#define LFUSE_DEFAULT (FUSE_SPIEN & FUSE_CKDIV8 & FUSE_SUT0 & FUSE_CKSEL0)
/* High Fuse Byte */
#define FUSE_RSTDISBL (unsigned char)~_BV(0)  /* Disable external reset */
#define FUSE_BODLEVEL0 (unsigned char)~_BV(1)  /* Enable BOD and select level */
#define FUSE_BODLEVEL1 (unsigned char)~_BV(2)  /* Enable BOD and select level */
#define FUSE_DWEN (unsigned char)~_BV(3)  /* DebugWire Enable */
#define FUSE_SELFPRGEN (unsigned char)~_BV(4)  /* Self Programming Enable */
#define HFUSE_DEFAULT (0xFF)

MCUdude, 2023-01-15:
If you need a stable 100kHz signal, this can be achieved using the internal timer. Note that the internal 9.6 MHz RC oscillator is usually a bit off, so you'll have two tweak the OCR0A/OCR0B values to get as close to 100kHz as possible. However, when using the timer, you'll stuck with pin PB0 and PB1.

```c
void setup() {

// Disable all interrupts
TIMSK0 |= (0<< OCIE0B) | (0<<OCIE0A) | (0<<TOIE0);

// Define outputs
DDRB = (1<<DDB4) | (1<<DDB3) | (1<<DDB2) | (1<<DDB1) | (1<<DDB0);

// Enable CTC
TCCR0A |= (1<<WGM01) | (0<<WGM00); 
TCCR0B |= (0<<WGM02);

// 9.6MHz / 100kHz (should be 43 if the internal oscillator were accurate)
OCR0A = 44;
OCR0B = 44;

// Toggle on Compare Match
TCCR0A |= (0<<COM0A1) | (1<<COM0A0) | (0<<COM0B1) | (1<<COM0B0);

// No prescaler
TCCR0B |= (0<<CS02) | (0<<CS02) | (1<<CS00);

// Make it start
GTCCR |= (1<<PSR10);

// Force one invert
TCCR0B |= (1<<FOC0B);
}

void loop() {
}
```

mcuee, 2023-05-12:
I think this issue can probably be closed.

closed 2023-05-12 by MCUdude