MCUdude/MicroCore #144: Known issues with M1 MacBook Pro?

jvlobo, 2023-02-11:
Hello!

Are there any known issues programming the ATTiny13 using Arduino IDE on a MacBook Pro M1?? I've been struggling quite a bit trying to upload some code but I'm getting errors all the time:

```
avrdude error: protocol expects OK byte 0x10 but got 0x14

avrdude error: protocol expects sync byte 0x14 but got 0x01
avrdude error: protocol expects sync byte 0x14 but got 0x10
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude error: unknown response=0x12
Failed programming: uploading error: exit status 1
```

Out of curiosity I tried on my older Intel MacBook Pro and it uploaded the code with no issues.

MCUdude, 2023-02-13:
Which programmer are you using? AFAIK there's no known issues with the M1 macs and Avrdude 7.1.
I only own an Intel mac, so I'm not able to test your setup, but I know @mcuee have access to an M1 mac.

MCUdude, 2023-02-13:
Can you turn on verbose upload in the IDE settings and paste the entire Avrdude output?
Thanks!

jvlobo, 2023-02-13:
Yes, this is what it shows:

```
avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/jvlobo/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/jvlobo/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1101
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware

avrdude stk500_getparm() error: protocol expects OK byte 0x10 but got 0x14

avrdude stk500_getparm() error: protocol expects sync byte 0x14 but got 0x02

avrdude stk500_getparm() error: protocol expects sync byte 0x14 but got 0x10
         Hardware Version: 32760
         Firmware Version: 48.48
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude stk500_initialize() error: protocol expects OK byte 0x10 but got 0x01
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude stk500_disable() error: protocol expects sync byte 0x14 but got 0x10

avrdude done.  Thank you.

Failed programming: uploading error: exit status 1
```

mcuee, 2023-02-13:
Yes I have Mac Mini M1. I should be able to carry out the test later (need to dig out my ATtiny13 breakout board).

On the other hand, I find it useful to get some types of `AVR ISP Shield` (very cheap from places like AliExpress) if you want to use `Arduino as ISP` more often. Usually I found ribbon cable based setup to be not as stable.

Ref: I was able to get 115200bps working with minor modification to the "Arduino as ISP" shield (remove all the LED handling codes).
* https://github.com/avrdudes/avrdude/discussions/1088

mcuee, 2023-02-14:
@jvlobo 

Doe the issue get better if you follow the prompt? That will require you to use the command line to run avrdude and not from within the Arduino IDE.
```
use -B to set lower ISP clock frequency, e.g. -B 125kHz
```

Do you have other AVR chips? If yes, do you have issues with other AVR chips as well?

mcuee, 2023-02-16:
No issues from my side.

```

mcuee@mcuees-Mac-mini avrdude_bin % uname -a
Darwin mcuees-Mac-mini.local 22.3.0 Darwin Kernel Version 22.3.0: Thu Jan  5 20:49:43 PST 2023; 
root:xnu-8792.81.2~2/RELEASE_ARM64_T8103 arm64

mcuee@mcuees-Mac-mini avrdude_bin % sw_vers
ProductName:		macOS
ProductVersion:		13.2
BuildVersion:		22D49

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude -c avrisp -P /dev/tty.usbmodem224301
 -b 19200 -p t13a -v

avrdude: Version 7.1-20230114 (5ec72e2)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
         User configuration file is /Users/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/tty.usbmodem224301
         Using Programmer              : avrisp
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude -c avrisp -P /dev/tty.usbmodem224301 -b 19200 -p t13a -U flash:r:t13a_blink_readback.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 0.80 s 

avrdude: writing output file t13a_blink_readback.hex

avrdude done.  Thank you.

```

mcuee, 2023-02-16:
Using Arduino IDE 2.0.3 and MicroCore.

```
Sketch uses 78 bytes (7%) of program storage space. Maximum is 1024 bytes.
Global variables use 0 bytes (0%) of dynamic memory, leaving 64 bytes for local variables. Maximum is 64 bytes.
"/Users/mcuee/Library/Arduino15/packages/MicroCore/tools/avrdude/7.1-arduino.1/bin/avrdude" "-C/Users/mcuee/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf" -v -V -pattiny13 -cstk500v1 -P/dev/cu.usbmodem224301 -b19200  -Uhfuse:w:0xfb:m -Ulfuse:w:0b00111010:m "-Uflash:w:/private/var/folders/zf/457r6vn50knfpnjgy1156llr0000gn/T/arduino-sketch-4DF5F3514B068F2C8A3A479ECA70D871/Blink.ino.hex:i"

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/mcuee/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem224301
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file 0xfb for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: reading input file 0b00111010 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: reading input file /private/var/folders/zf/457r6vn50knfpnjgy1156llr0000gn/T/arduino-sketch-4DF5F3514B068F2C8A3A479ECA70D871/Blink.ino.hex for flash
         with 78 bytes in 1 section within [0, 0x4d]
         using 3 pages and 18 pad bytes
avrdude: writing 78 bytes flash ...

Writing | ################################################## | 100% 0.21s

avrdude: 78 bytes of flash written

avrdude done.  Thank you.

```

mcuee, 2023-02-16:
@jvlobo 
Just wondering is the issue is specific to certain sketech or it is the same with the default Arduino blink sketech.

If if is specific sketch and if it is okay to share the sketch, please share the sketch and I can carry out the test again. I am using the AVR ISP shield with an Arduino Uno clone to form the "Arduino as ISP" programmer in this case.

jvlobo, 2023-02-16:
Hey @mcuee thanks a lot for trying it on your side. I'll try with the command line as soon as possible.
I was using my own sketch but then for testing I used the blink example from the Arduino IDE. But I've just tried now with the blinking sketch and it worked 😰. That is strange... I would swear the blinking sketch didn't work for me when I was trying, but I tested and changed so many things that who knows now....

My own sketch is still not uploading for some reason (it's compiling with no issues tho).

This is my sketch: 

```
const uint8_t LEDS_TABLE[8][4] = {
  {0, 1, 2, 2},
  {1, 0, 2, 2},
  {2, 0, 1, 2},
  {2, 1, 0, 2},
  {2, 2, 0, 1},
  {2, 2, 1, 0},
  {1, 2, 2, 0},
  {0, 2, 2, 1},
};
const int PINS[4] = {2, 3, 4, 1};

void setup() {}

void light_led(int led_number) {
  for(int i = 0; i < 4; i++) {
    if(LEDS_TABLE[led_number - 1][i] == 2) {
      pinMode(PINS[i], INPUT);
    } else if(LEDS_TABLE[led_number - 1][i] == 0) {
      pinMode(PINS[i], OUTPUT);
      digitalWrite(PINS[i], LOW);
    } else {
      pinMode(PINS[i], OUTPUT);
      digitalWrite(PINS[i], HIGH);
    }
  }
  
  delay(500);
}

void loop() {
  for(int i = 1; i <= 8; i++) {
   light_led(i);
  }
}
```

I would say the only "weird" thing I'm using is the `uint8_t` type. But this uploads and works fine when doing it from my Intel computer so no idea what is happening.

Again, thank you for your kind help.

mcuee, 2023-02-19:
@jvlobo 

I tried your sketch and there is no issue on my side to upload and run.

```
Sketch uses 208 bytes (20%) of program storage space. Maximum is 1024 bytes.
Global variables use 40 bytes (62%) of dynamic memory, leaving 24 bytes for local variables. Maximum is 64 bytes.
"/Users/mcuee/Library/Arduino15/packages/MicroCore/tools/avrdude/7.1-arduino.1/bin/avrdude" "-C/Users/mcuee/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf" -v -V -pattiny13 -cstk500v1 -P/dev/cu.usbmodem224201 -b19200  -Uhfuse:w:0xfb:m -Ulfuse:w:0b00111010:m "-Uflash:w:/private/var/folders/zf/457r6vn50knfpnjgy1156llr0000gn/T/arduino-sketch-C995F7364482E0F552A45FF202899C3F/sketch_feb19a.ino.hex:i"

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/mcuee/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem224201
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file 0xfb for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: reading input file 0b00111010 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: reading input file /private/var/folders/zf/457r6vn50knfpnjgy1156llr0000gn/T/arduino-sketch-C995F7364482E0F552A45FF202899C3F/sketch_feb19a.ino.hex for flash
         with 208 bytes in 1 section within [0, 0xcf]
         using 7 pages and 16 pad bytes
avrdude: writing 208 bytes flash ...

Writing | ################################################## | 100% 0.47s

avrdude: 208 bytes of flash written

avrdude done.  Thank you.
```

mcuee, 2023-02-19:
@jvlobo 

I tend to think your connections may be marginal. You may want to try the command line and use `-B` options to see if that helps.

jvlobo, 2023-02-22:
Thanks again @mcuee for your kind help and sorry for the delay.

I've tried using the terminal with `avrdude` and I'm actually getting the same issue that in Arduino... no idea why 🤔

```
    ~  avrdude -c avrisp -P /dev/cu.usbmodem1101 -b 19200 -p t13a -v                                                                                            

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/opt/homebrew/etc/avrdude.conf"
         User configuration file is "/Users/jvlobo/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1101
         Using Programmer              : avrisp
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP

avrdude: stk500_getparm(): (a) protocol error, expect=0x10, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 1
         Firmware Version: 1.696387080
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.
```

jvlobo, 2023-02-22:
Nevermind.... I tried again and it worked, I was missing something.
I tried also on Arduino IDE now and it worked 😰😰🤯🤯

Literally no idea what happened the previous times... 🤦🏽‍♂️ but yeah definitely not an issue with newer Apple devices.

Thanks a lot again guys for the help 🙏🏽

closed 2023-02-22 by jvlobo