MCUdude/MicroCore #146: Calling noTone() may leave pin high, drawing current

pserwylo, 2023-04-01:
# Details

When calling `noTone()`, there is a 50% chance that we accidentally leave a speaker fully energised, and hence drawing current.
This depends on whether the wave was high or low at the time we called `noTone()`.

Not only does it draw current, but it will be twice the current compared to when the tone was playing, because
the tone only drives the pin at 50% duty cycle, whereas leaving the pin high after `noTone()` causes twice this current draw.

By forcing the pin low when calling `noTone()`, we make this deterministically not draw current afterwards.

# Proof of Concept

The proof of concept is deterministic, in that I've chosen a time to call `noTone()` that almost always (on my attiny13a at least)
seems to cause high current draw after `noTone()`. However my actual use case is to call `noTone()` in response to user input, which
will randomly either be in the high or low portion of the currently playing tone, and thus 50% of the time result in high current
draw after a call to `noTone()`.

## Observations

In the proof of concept, I've observed the following when running at 9.6Mhz:
 * 4.5ma when idle.
 * 37.5ma when playing the tone.
 * 65-70ma after calling `noTone()`.

## Code

```

#define SPEAKER_PIN PB4

/**
 * Using a nice rounded frequency of 100Hz (10ms per wave, 5ms high and 5ms low at 50% duty cycle)
 * to aid in calculating when the wave is likely to be high vs when the wave is likely to be low.
 */
#define TONE_FREQ 100

/**
 * Likely to be in the high section of a wave after this duration. Hence, if we call noTone(), it will
 * still be sending voltage to the speaker, and thus it will still be sinking current. Furthermore, it
 * will be sinking twice the current as compared to when the speaker is playing, because it is holding
 * the magnet in an energised state for the full duty cycle when fixed high.
 */
#define PROBLEMATIC_ON_DURATION_MS 2008

/**
 * Likely to be in the low section of a wave.
 */
#define UNPROBLEMATIC_ON_DURATION_MS 2000

#define OFF_DURATION_MS 2000

void setup() {
  pinMode(SPEAKER_PIN, OUTPUT);
}

void loop() {
  // Expect this to draw a lot of current after calling noTone().
  for (int i = 0; i < 5; i ++) {
    tone(SPEAKER_PIN, TONE_FREQ);
    delay(PROBLEMATIC_ON_DURATION_MS);
    noTone(SPEAKER_PIN);
    delay(OFF_DURATION_MS);
  }

  // Expect this to draw no additional current after calling noTone().
  for (int i = 0; i < 5; i ++) {
    tone(SPEAKER_PIN, TONE_FREQ);
    delay(UNPROBLEMATIC_ON_DURATION_MS);
    noTone(SPEAKER_PIN);
    delay(OFF_DURATION_MS);
  }

  // Even though this is using a PROBLEMATIC_ON_DURATION_MS, it will not draw any additional current
  // after calling noTone().
  for (int i = 0; i < 5; i ++) {
    tone(SPEAKER_PIN, TONE_FREQ);
    delay(PROBLEMATIC_ON_DURATION_MS);
    noTone(SPEAKER_PIN);
    digitalWrite(SPEAKER_PIN, LOW);
    delay(OFF_DURATION_MS);
  }
}
```

pserwylo, 2023-04-01:
For my use case, calling digitalWrite(LOW) explicitly is fine. However if you think my analysis above is correct (which it may not be - I'm a software engineer starting my electronics hobby journey and so often get things wrong!) it may be worth adding the equivalent at the end of the noTone implementation.

pserwylo, 2023-04-02:
I can clarify that after testing, it doesn't exhibit the same problem with a call to `tone()` that specifies a duration, if we just leave the tone to naturally get to the end of its duration.

MCUdude, 2023-04-10:
Thanks for the comprehensive testing and explanation! I've just pushed a fix, and it will be available in the next MicroCore boards manager release.

Thank you for using MicroCore"