MCUdude/MicroCore #147: No programmer on Arduino 2

kamilbystryk, 2023-04-03:
On Arduino 1.x everything it's OK, but on Arduino 2 I don't see "Arduino as ISP (micro code)" programmer. In this case, when I choose "Arduino asISP" i can only write bootloader, but I can't write program. How can I fix it?
[image]

MCUdude, 2023-04-10:
Have you installed MicroCore using the boards manager install? I have no issues running "Burn Bootloader" from IDE 2.0.4

[image]

mcuee, 2023-05-12:
No issues here using Board Manager with latest Arduino IDE 2.1.0 under Windows 11.

mcuee, 2023-05-19:
I think this issue can be closed.

closed 2023-05-19 by MCUdude