MCUdude/MicroCore #148: tone() not working

kratochviljan, 2023-04-05:
I'm trying the basic wiring and the tone() function, but the only thing that happens is that the pin goes HIGH.
(checked with an oscilloscope)
I've tried different pins, different frequencies, different internal oscillator settings, always the same.
Does tone() function work for anyone ?
([Microtone](https://github.com/Master811129/MicroTone) for same wiring works fine)

Arduino 1.8.10, Vcc is single Li-ion cell 4.1V, 

[image]

[image]

MCUdude, 2023-04-10:
You can't repedeatly call `tone()`, as this will restart the timer every time.

Try this instead (just a proof of concept, no button debouncing added):

```c
#define BUZZER 0
#define BUTTON 1

void setup() {
  pinMode(BUZZER, OUTPUT);
  pinMode(BUTTON, INPUT_PULLUP);
}

void loop() {
  static bool prev_state;
  
  if(digitalRead(BUTTON) == LOW && prev_state == HIGH) {
    tone(BUZZER,440);
    prev_state = LOW;
  } 
  if(digitalRead(BUTTON) == HIGH && prev_state == LOW) {
    noTone(BUZZER);
    digitalWrite(BUZZER, LOW);
    prev_state = HIGH;
  }
}
```

closed 2023-04-13 by MCUdude

kratochviljan, 2023-04-16:
Yes, all clear, my stupidity.
(but I'm sticking with Microtone(), because Tone() with the rest of the code won't fit in 1024B Flash)