MCUdude/MicroCore #149: insomnia - if millis() is used it does not enter sleep mode

kratochviljan, 2023-04-08:
If the millis() function is used then it is not possible to enter sleep mode.

Interestingly, even if the sleep is done right at the beginning in the setup() function and the millis() is used later in the function that is called from the main loop(), it still fails to enter the sleep mode.

Just comment out line 19 - the call to killsleep(), which uses the millis() function, and everything works correctly.
The MCU goes into sleep mode immediately after reset, where it stays until the button is pressed.

[image]

```
#include <avr/sleep.h>
#include <avr/interrupt.h>

#define ledPin 0
#define buttonPin 1

void setup() {
  
  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, LOW);
  pinMode(buttonPin, INPUT_PULLUP);

  mcuSleep();
}

void loop() {

  blink();
  killsleep(); // comment this line  
}

void mcuSleep(){
  sei();
  sleep_enable();
  attachInterrupt(digitalPinToInterrupt(buttonPin), wakeInterrupt, LOW);
  delay(100);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_mode();
  //sleep here
  sleep_disable();
  delay(100);
  detachInterrupt(digitalPinToInterrupt(buttonPin));
}

void wakeInterrupt(void){
// NOP
}

void blink(void) {

    digitalWrite(ledPin, HIGH);
    delay(1000);
    digitalWrite(ledPin, LOW);
    delay(1000);  

}

void killsleep (void) {

  uint32_t now = millis();
  
}
```

MCUdude, 2023-04-08:
millis() runs off the internal watchdog timer. Can you try to disable watchdog interrupts before entering sleep mode?

```c
WDTCR &= ~(1 << WDTIE);
```

pklosko, 2023-04-08:
Hello Jan,  
my sleep funcion work fine @ attint13 :

int sleepIn(void) {
  if (DDRB != 0){
    ddr = DDRB;
  }
  if (DIDR0 != 0x3F){
    dir = DIDR0; 
  }
  DDRB = 0;                            // All lights off
  PORTB = 0x1F;                        // Pullups on
  DIDR0 = 0x3F;                        //Disable digital input buffers on all unused ADC0-ADC5 pins.
  ADCSRA &= ~(1<<ADEN);                //Disable ADC
  ACSR = (1<<ACD);                     //Disable the analog comparator
  cli();                               //Deactivate Interrupts as long as I change Bits      
  GIMSK |= (1<<INT0);                  // Enable Pin Change Interrupts
  set_sleep_mode(SLEEP_MODE_PWR_DOWN); // go to deep sleep
  MCUCR &= ~(1 << ISC00);              //Set low level Interrupt
  MCUCR &= ~(1 << ISC01);              //Set low level Interrupt
  sei();                               //Activate Interrupts
 //go to sleep
  sleep_mode();
  //and wake up
  cli();          //Deactivate Interrupts
  return 1;
}

kratochviljan, 2023-04-09:
@pklosko - there is no problem with the sleep routine, it works fine if the millis() function is not used

@MCUdude  - thank you, disabling watchdog interrupts solves the problem.
After waking up you have to turn it on again, otherwise millis() doesn't work, but that's obvious...
I'm listing here my sleep() function, which puts the mcu to sleep and then wakes it up by pressing a button, maybe that will help someone. 

[image]

```
void mcuSleep() {

  // because of the implementation of the millis() function it is necessary
  // to disable the watchdog interrupt before entering sleep mode
  WDTCR &= ~(1 << WDTIE);
   
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  noInterrupts(); // for the sake of keeping exact timing
  sleep_enable();
  attachInterrupt(digitalPinToInterrupt(buttonPin), wakeInterrupt, LOW);
  interrupts(); // enable interrrupts again
  sleep_cpu(); // here the cpu is put to sleep
  //Chrrrr, chrrrr
  sleep_disable(); // and here he wakes up again
  delay(100);
  detachInterrupt(digitalPinToInterrupt(buttonPin));

  // after waking the cpu from sleep mode it is necessary to enable
  // the watchdog interrupt again, otherwise the millis() function does not work
  WDTCR |= (1 << WDTIE);
}
```

MCUdude, 2023-04-10:
Great to hear you figured it out!

closed 2023-04-10 by MCUdude