MCUdude/MicroCore #150: Cant burn bootloader using 127khz internal oscilator

darkmetalman, 2023-05-08:
Hello!
Im trying to burn he bootloader with 127KHz internal oscilator selected and using "Arduino as ISP slow" programmer but it fails and results in an unusable attiny13 (wich can be restored using a high voltage programmer).

Is there something else to take into account before trying to use a 127KHz int osc?

I am using arduino ide 2.1.0, does it have compatibility issues?

Thanks!!

MCUdude, 2023-05-09:
Hi!

Can you turn on verbose Upload in the IDE settings and post the Avrdude output when you're trying to burn the bootloader?
Also, can you share the settings you use in the Arduino IDE Tools menu?

MCUdude, 2023-05-09:
I had a look at the Arduino as ISP sketch. This programmer is very primitive, and I never use this myself because it's slow, error-prone and you'll have to DIY it, which often leads to bad connections.

Personally, I prefer to use the USBasp or USBtinyISP programmers.

What you'll probably have to do is to modify the Arduino as ISP sketch to lower the ISP clock speed, as it seems like it doesn't support this using Avrdude.

See this part in the sketch:

```c
// Configure SPI clock (in Hz).
// E.g. for an ATtiny @ 128 kHz: the datasheet states that both the high and low
// SPI clock pulse must be > 2 CPU cycles, so take 3 cycles i.e. divide target
// f_cpu by 6:
//     #define SPI_CLOCK            (128000/6)
//
// A clock slow enough for an ATtiny85 @ 1 MHz, is a reasonable default:

#define SPI_CLOCK 		(1000000/6)

```

darkmetalman, 2023-05-09:
Thank you MCUdude, i'll take your advice and get one of the programmers that you mentioned.