MCUdude/MicroCore #151: Optimize shiftOut function a little

marshfolx, 2023-05-09:
I think the value of MSBFIRST and LSBFIRST should be changed to 0x80 and 0x01 respectively. Then a reduntant branch can be elimated to reduce code size. I have no idea if this has any side effect to existed code.

From this:

```cpp
    if(bitOrder == MSBFIRST)
    {
      if(value & 0x80)
        PINB = datapinMask;
    }
    else
    {
      if(value & 0x01)
        PINB = datapinMask;
    }
```

To:

```cpp
      if(value & static_cast<uint8_t>(bitOrder))
        PINB = datapinMask;
```

MCUdude, 2023-05-09:
Hi @marshfolx!

Thanks for the suggestion! I haven't tried your code on actual hardware, but it seems like the compiler is clever enough to optimize the existing shiftOut function so that the compiled size is identical to the improvement you provided.

The compiler probably understands that the MSBFIRST/LSBFIRST argument is constant, and optimizes away the entire if-query.

There might be code out there that relies on `MSBFIRST == 1` and `LSBFIRST == 0`, so I'm a bit hesitant to change these.

marshfolx, 2023-05-21:
@MCUdude 
emmm，if so then I did agian some premature optimizations😂. I'm too lazy to find out the truth in assembly.

closed 2023-05-21 by marshfolx