MCUdude/MicroCore #152: Problems with firmware

Bogdan2282, 2023-08-03:
I filled a sketch on Attiny13A and accidentally set the clocking frequency to 16Mhz. After that I can't flash my MC anymore. How can I flash it. I have usbasp and aduino nano. Can I fix my mc

MCUdude, 2023-08-03:
You need to apply a clock signal to the ATtiny13 in order to recover it. It can't drive a crystal on its own, but if you use your Arduino Nano you can flash [this modified Arduino as ISP sketch](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/ArduinoISP/ArduinoISP/ArduinoISP.ino) that outputs a steady 8 MHz clock signal on digital pin 9. Connect pin 9 to pin 3 (PB3 on the ATtiny13, and you should be able to recover it using your USBasp.

Bogdan2282, 2023-08-03:
Everything worked out. Thank you very much. What would I do without you.

MCUdude, 2023-08-04:
You're welcome!

closed 2023-08-04 by MCUdude