MCUdude/MicroCore #154: Attiny10 support

Peterhj05, 2023-11-16:
Hi.
Love it! Well done!
Can it support attiny10 by ignoring 2 io pins and modify the circuit to suit?
Kind regards
Peter

MCUdude, 2023-11-16:
Hi!

Sorry, but I'm not planning to add ATtiny10 support.
The chip is fundamentally different with way fewer instructions (54 instead of 120 as the ATtiny13 has), so the source code must be rewritten to end up with core functionality that isn't bloated.

You can try [this core](https://github.com/technoblogy/attiny10core) which lets you use Arduino IDE for programming but does not support any "Arduino functionality" like pinMode and digitalWrite. Or you have [this core](https://github.com/sleemanj/ATTinyCore) which provides _some_ Arduino functionality.

closed 2023-11-16 by MCUdude

Peterhj05, 2023-11-17:
Hi. Apprecite the reply and the links. Thank you. Peter