MCUdude/MicroCore #155: millis() prevents entering to sleep

lazarevd, 2024-01-25:
If millis() function appears anywhere in loop(), it prevents enetering to sleep mode. The position of millis() doesn`t matter, it could be after  sleep_cpu(). 
micros() works normally.

`#include <avr/sleep.h>
#include <avr/interrupt.h>
#include <avr/power.h>
#define LED PB0

void setup() {
  pinMode(LED, OUTPUT);
  delay(500);
}

void loop() {
  analogWrite(LED, 0);
  delay(200);
  analogWrite(LED, 250);
  delay(200);
  analogWrite(LED, 0);
  system_sleep();
  millis();//commenting this line let attiny13 enter to sleep, if i uncomment it, led blinks continously
}

void system_sleep() {
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  sei();
  sleep_cpu();
}`

radiantwave, 2024-08-29:
[millis()](https://www.arduino.cc/en/Reference/Millis) in the MicroCore are watchdog timer based.
Do you still need help with this?

lazarevd, 2024-08-30:
Thank you. After all I used micros() instead of millis(), but i’ll try your solution. 
Is it in documentation? May be it’s worth to add it if it’s not.