MCUdude/MicroCore #159: What is the default Vref value set at for ATtiny13A?

VidhuSarwal, 2024-09-08:
I want to add a battery level detector to the ATtiny13A, where the same battery powers the microcontroller. My approach is to use ADC Pin 1 (PB2) with a voltage divider to detect the battery voltage.

**Key Questions:**
1. Is the default Vref set to 1.1V on the ATtiny13A? I plan to use this internal reference for voltage comparison.
2. I came across the [ArduinoSystemStatus library](https://github.com/cano64/ArduinoSystemStatus), which can check VCC levels. Will this library be compatible with the ATtiny13A for this purpose?

_If you have any other recommendations for detecting the battery voltage, please let me know._

MCUdude, 2024-09-08:
Hi!

The default Vrevf is the supply voltage. The only alternative is to use the internal 1.1V reference.

https://github.com/MCUdude/MicroCore/blob/762f91ff30f4679584e08807cedc728682ce0c77/avr/cores/microcore/wiring_analog.c#L24-L30

It doesn't seem like the library you're referring to is designed to work with the ATtiny13. I'm not all that familiar with low voltage operation, since the electronics I usually design are always connected to a power supply of some sort. But you can try playing around with the [avrlibc sleep.h library](https://www.nongnu.org/avr-libc/user-manual/group__avr__sleep.html).

pklosko, 2024-09-11:
See example in my code >
https://github.com/pklosko/attiny13_thermoE32-868/blob/master/attiny13_thermoE32-868.ino#L119

VidhuSarwal, 2024-09-12:
Thanks @MCUdude & @pklosko for your help. 

I am closing this.

closed 2024-09-12 by VidhuSarwal