MCUdude/MicroCore #160: -ffixed is no longer needed?

WestfW, 2024-09-25:
platform.txt contains:
```
# fixed registers: r4 = isr_tmp, r3:r2 = WDT counter for millis
#compiler.optimize_flags=-Os -flto -fno-move-loop-invariants -fno-reorder-blocks -ffixed-r2 -ffixed-r3 -ffixed-r4
compiler.optimize_flags=-Os -flto -fno-reorder-blocks -ffixed-r2 -ffixed-r3 -ffixed-r4

```
However, it looks like the nerdRalph millis code no longer keeps stuff in registers; I can't find any actual used of r2-r4 anywhere in the current source code.

(it is unlikely that this will change anything in any program small enough to fit in a tiny13, but...)

MCUdude, 2024-09-25:
To be honest, I don't know for sure. I was struggling to get the size of a compiled, blank sketch down to the same size as @nerdralph's [picoCore](https://github.com/nerdralph/picoCore) would achieve. I asked him what the secret sauce was, and I was allowed to borrow code (and platform.txt data) to achieve similar results.

Can you try to remove the -ffixed flags and see how/if it makes a difference?

WestfW, 2024-09-25:
It looks like picoCore uses an entirely register-based WDT/millis implementation: https://github.com/nerdralph/picoCore/blob/master/avr/cores/picocore/millis.S#L23
While microCore has the somewhat bigger NerdRalph Memory-based millis() (when it's turned on.) ( https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/millis.S#L46 )

MCUdude, 2024-09-25:
Not sure why I haven't caught the change. Would you like to submit a PR for it? I'm all for saving every bit of memory on these tiny AVRs!

nerdralph, 2024-09-25:
The discussion was years ago, but my recollection is that you opted for the
larger code size for better accuracy.  Your sample of t13s had an average
WDT frequency around 110 kHz, and chose a 19/16 correction factor.
PicoCore supports the t13 and t85, and my sampling of chips had a somewhat
higher average WDT frequency, and minimum size took priority over accuracy.

closed 2026-02-10 by MCUdude