MCUdude/MicroCore #161: ATtiny10 support?

WestfW, 2024-09-25:
I've noticed a couple people trying to use ATtiny10s recently.  They have as much flash as an ATtiny13, so it's ... not impossible.
There is at least on "core", but it turns out that the Arduino compiler currently has something wrong with it, WRT avrtiny, so you have to use an old version of the IDE (1.8.9, before Arduino started using their own build of v7.3.  See https://www.avrfreaks.net/s/topic/a5CV400000027bhMAA/t397960 )

In addition, those cores seem to be abandoned, and not particularly well optimized.
So I decided to look at what it would take to add tiny10 support to microcore.  The needed changes look pretty small; mostly WRT changes in interrupt control registers and such.

Are you interested in this?  (there's still the "compiler is broken" problem, though.)

MCUdude, 2024-09-25:
> Are you interested in this? (there's still the "compiler is broken" problem, though.)

Absolutely! However, it is that broken compiler thing though. You'll have to use Arduino's avr-gcc 5.4.0 for it to work, but that can be sorted out by using the boards manager when installing. But that would mean downgrading from 7.3.0 when compiling for the ATtiny13. Unless we can pull off some platform.txt trickery that would allow us to switch between compilers, and install both when installing MicroCore through boards manager.

There are lots of resources here, and AFAIK is pretty well optimized for the ATtiny4/5/9/10s.
https://github.com/sleemanj/ATTinyCore

I'm all for this, but keep in mind that my spare time is quite limited. I also want to maintain the same "feel" as the core has today. Unlike other cores for small AVR microcontrollers, I favor a straight-to-the-point README, a Tools menu option that isn't bloated with lots and lots of settings a pro user doesn't really care about, and clever things like the SW UART implementation and automatically optimizing out millis and micros if not used in the user program.

And if there are things that the T9/10 cannot support, we should be very clear about what that is in the README.

WestfW, 2024-09-25:
Oh!  I hadn't realized that ATtinyCore added avrtiny chips (they're not in the latest "board manager" version (1.5.2, I think.  Vs 2.x on github.)

> You'll have to use Arduino's avr-gcc 5.4.0 for it to work

It's weirder than that.  Microchip's distributed version of 7.3 works fine.