MCUdude/MicroCore #162: Problems disabling safe mode.

XanCraft21, 2024-11-14:
Boards version: latest (i will add the version number later),
Arduino version: 1.8.19,
Operating system: windows 10,

Hello, i am having trouble disabling safe mode in this software package, my sketch doesn’t want to fit, and it’s not even that big (it is a little but i thought it would fit). According to the description in the core_settings h file, if i disable it, it will give me more flash space, and i can see that the flash usage seems to ramp up quickly, even for a small code.

Unfortunately disabling safe mode has the opposite effect, taking up 2 single bytes of flash instead of giving me a bunch more free space. I tried commenting out the define for it, i tried defining DISABLE_SAFEMODE (i think i have it right) in my main sketch like i saw where it says ifndef in core_settings, i can’t disable it correctly.

Is this a bug in the most recent release? Thank you for any help you can give me.

MCUdude, 2024-11-14:
Can you post your sketch? There may be other things you can do to optimize your code to make it fit.

MCUdude, 2024-11-14:
I'll look into why the sketch size increases.

XanCraft21, 2024-11-15:
I will post my sketch in a gist when possible. I was using the standard arduino functions, so maybe that’s contributing?

I could always try avr c code like shown in some of the examples, which will require me to make some of the gpio functions myself.

Do you know of another way i can disable safe mode to get more flash space?

XanCraft21, 2024-11-26:
Hello, i wanted to post a quick update on this issue.

I noticed that this boards package has gotten updated. After installing it, regarding the safe mode thing, not a big change has happened. Disabling it no longer adds 2 bytes of flash usage, but i still don’t see anything decreasing when i do disable it.

As for the sketch i got a bit sidetracked, so i will post that when i can.