MCUdude/MicroCore #163: MicroCore doesn't work after update Arduino IDE for ver. 2.3.4

Denexp-1, 2024-12-05:
When I'm trying to load bootloader to ATtiny13 have errors:

Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\Users\xxxxxxx\AppData\Local\Arduino15\packages\MicroCore\tools\avrdude\8.0-arduino.1\etc\avrdude.conf

Using port            : COM37
Using programmer      : stk500v1
Setting baud rate     : 19200
Setting bit clk period: 32.0 us

Error: cannot get into sync
Error: cannot set Parm_STK_SCK_DURATION
Error: unable to open port COM37 for programmer stk500v1

Avrdude done.  Thank you.
Failed chip erase: uploading error: exit status 1

JoergCH, 2024-12-20:
I can confirm that there is problem. It has (probably) nothing to do with the Arduino version but (probably) with the update of MicroCore to to 2.4.1, or to a related update in Avrdude (from 7.2 to 8.0). 

On a system with Arduino 2.3.2 and MicroCore 2.4.0, I have no issues at all.

On another system with Arduino 2.3.4 and MicroCore 2.4.1, I have issues when I try to burn a bootloader:

    Avrdude version 8.0-arduino.1
    Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

    System wide configuration file is /home/xxx/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
    User configuration file /home/xxx/.avrduderc does not exist

    Using port            : /dev/ttyUSB0
    Using programmer      : stk500v1
    Setting baud rate     : 19200
    Setting bit clk period: 32.0 us

    Error: cannot get into sync
    Error: cannot set Parm_STK_SCK_DURATION
    Error: unable to open port /dev/ttyUSB0 for programmer stk500v1

    Avrdude done.  Thank you.
    Failed chip erase: uploading error: exit status 1

On that same system, I kept Arduino 2.3.4 but reverted back to MicroCore 2.4.0. Now everything works:

    avrdude: Version 7.2-arduino.1
            Copyright the AVRDUDE authors;
            see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

            System wide configuration file is /home/xxx/.arduino15/packages/MicroCore/hardware/avr/2.4.0/avrdude.conf
            User configuration file is /home/xxx/.avrduderc
            User configuration file does not exist or is not a regular file, skipping

            Using Port                    : /dev/ttyUSB0
            Using Programmer              : stk500v1
            Overriding Baud Rate          : 19200
            Setting bit clk period        : 32.0
            AVR Part                      : ATtiny13A
            Chip Erase delay              : 4000 us
            RESET disposition             : dedicated
            RETRY pulse                   : SCK
            Serial program mode           : yes
            Parallel program mode         : yes
            Timeout                       : 200
            StabDelay                     : 100
            CmdexeDelay                   : 25
            SyncLoops                     : 32
            PollIndex                     : 3
            PollValue                     : 0x53
            Memory Detail                 :

                                              Block Poll               Page                       Polled
              Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
              ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
              eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
              flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
              lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
              hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
              lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
              signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
              calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

            Programmer Type : STK500
            Description     : Atmel STK500 version 1.x firmware
            Hardware Version: 2
            Firmware Version: 1.18
            Topcard         : Unknown
            Vtarget         : 0.0 V
            Varef           : 0.0 V
            Oscillator      : Off
            SCK period      : 0.1 us
    avrdude: AVR device initialized and ready to accept instructions
    avrdude: device signature = 0x1e9007 (probably t13a)
    avrdude: erasing chip

    avrdude: processing -U lock:w:0xff:m
    avrdude: reading input file 0xff for lock
            with 1 byte in 1 section within [0, 0]
    avrdude: writing 1 byte lock ...
    avrdude: 1 byte of lock written
    avrdude: verifying lock memory against 0xff
    avrdude: 1 byte of lock verified

    avrdude: processing -U hfuse:w:0xeb:m
    avrdude: reading input file 0xeb for hfuse
            with 1 byte in 1 section within [0, 0]
    avrdude: writing 1 byte hfuse ...
    avrdude: 1 byte of hfuse written
    avrdude: verifying hfuse memory against 0xeb
    avrdude: 1 byte of hfuse verified

    avrdude: processing -U lfuse:w:0b00111010:m
    avrdude: reading input file 0b00111010 for lfuse
            with 1 byte in 1 section within [0, 0]
    avrdude: writing 1 byte lfuse ...
    avrdude: 1 byte of lfuse written
    avrdude: verifying lfuse memory against 0b00111010
    avrdude: 1 byte of lfuse verified

    avrdude done.  Thank you.

My suggestion for a workaround: revert back to 2.4.0.

Denexp-1, 2024-12-20:
OK I'll check it with WA. Thanks a lot for your investigation and answer!

Denexp-1, 2024-12-20:
Yes, it works Arduino IDE 2.3.4 and MicroCore 2.4.0.

F8RDev, 2025-03-09:
Doesn't work again in MicroCore 2.4.1.

JoergCH, 2025-05-04:
Update: The problem persists also with MicroCore 2.5.1 and Arduino IDE 2.3.6.
The workaround of using MicroCore 2.4.0 works with Arduino IDE 2.3.6.

MCUdude, 2025-05-04:
I'd really like to get to the bottom of this. It appears to be an issue with Avrdude when using Arduino as ISP with a baud rate of 19200 baud. I find this very strange, because there haven't been any major changes to the Avrdude codebase between Avrdude 7.2 (used by MicroCore 2.4.0) and Avrdude 8.0 (used by MicroCore 2.4.1).

Avrdude 8.1 is soon to be released, and maybe, just maybe, we can isolate the issue and submit a patch before the 8.1 release.

For all of you who experience issues with Arduino as ISP, I need to know the following:
* What kind of OS are you running?
* Which IDE version are you using?
* What kind of USB to serial chip is used between the microcontroller running the Arduino as ISP sketch and the computer?

mcuee, 2025-05-04:
@JoergCH 
Which Arduino Nano are you using?

Official one will be using FTDI FT232RL USB to TTL chip.
https://store.arduino.cc/products/arduino-nano

Many clones will using things like CH340.

Is it possible that you build avrdude from source and then see if it has the same issue or not. Thanks.

MCUdude, 2025-05-04:
> @JoergCH do you have a 10uF capacitor between the reset pin and GND on your Arduino Nano (running the Arduino as ISP sketch)? This is necessary to prevent the auto reset functionality from disturbing the programming.

> Is it possible that you build avrdude from source and then see if it has the same issue or not. Thanks.

Oh, that would be wonderful! If @JoergCH could build the latest Avrdude git main to figure out what we can do about this issue (I'm involved in the Avrdude development BTW)

JoergCH, 2025-05-04:
The Nano is a clone. Chip claims to be a CH340C, processor a Mega328P. Photos attached :-)

@MCUdude - Hans, I'm willing to do that but I would appreciate a few pointers how I do that _without_ "disturbing" the rest of the machine. In particular, I would need to know where exactly I can tell the Arduino IDE what binary it should use ;-) (yes, I'm running everything from inside the IDE).

[image]
[image]
[image]

MCUdude, 2025-05-04:
@JoergCH It looks to me like there is a capacitor (10uF or more?) connected between reset and ground. 

> @MCUdude - Hans, I'm willing to do that but I would appreciate a few pointers how I do that without "disturbing" the rest of the machine. In particular, I would need to know where exactly I can tell the Arduino IDE what binary it should use ;-) (yes, I'm running everything from inside the IDE).

Don't worry! MicroCore (and all my other cores) bundles its own version of Avrdude to make sure that it will work on any system. If you open your terminal and run `$ avrdude`, you'll probably get a command not found error.

Clone the Avrdude repo and follow the instructions found here:  
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux

This will install Avrdude on your system, but will not interfere with the Arduino IDE. In order to test Avrdude you run it directly from your terminal, not through Arduino IDE. For instance, you can run this command, which reads the contents of the flash memory:

`avrdude -c stk500v1 -P /dev/ttyUSB0 -b 19200 -p attiny 13 -U flash:r:-:I`

stefanrueger, 2025-05-04:
It could well be that at some point programmers were renamed to get on top of the (avrdude) guessing game which protocol variant was needed (an otherwise impossible to solve problem whether flash/eeprom addresses were in words or bytes).
```
$ avrdude -c*/d | grep stk500[^a-z]
    'avrisp' =>              ['stk500', 'PM_ISP', 'Serial Atmel AVR ISP using STK500'], # /usr/local/etc/avrdude.conf 1497
    'stk500' =>              ['stk500generic', 'PM_ISP', 'Atmel STK500 (probes v2 first then v1)'], # /usr/local/etc/avrdude.conf 1594
    'stk500v1' =>            ['stk500', 'PM_ISP', 'Atmel STK500 v1'], # /usr/local/etc/avrdude.conf 1611
    'arduino_as_isp' =>      ['stk500', 'PM_ISP', 'AVR as programmer with Arduino-as-ISP FW'], # /usr/local/etc/avrdude.conf 1629
    'mib510' =>              ['stk500', 'PM_TPI | PM_ISP', 'Crossbow MIB510 programming board'], # /usr/local/etc/avrdude.conf 1647
```

As you can see, nowadays `-c stk500v1` *only* caters for the Atmel STK500 programmer with v1 firmware; there are different instantiations for those programmers that utilise the same (well, that was the problem: a very similar but not the same) protocol. Try `-c avrisp` or (more likely?) `-c arduino_as_isp`.

mcuee, 2025-05-05:
> As you can see, nowadays `-c stk500v1` _only_ caters for the Atmel STK500 programmer with v1 firmware; there are different instantiations for those programmers that utilise the same (well, that was [...]

`-c arduino_as_isp` is the way to go in this case. But I remember the main difference is the EEPROM handling.

`-c avrisp` is supposed to be the same as `-c stk500v1`.

`arduino_as_isp` is the same as optiboot in terms of EEPROM handling, but different from official `stk500v1`.
https://github.com/avrdudes/avrdude/blob/58408592b99608b93ae62ac16525634b3ef9c047/src/stk500.c#L979-L985

mcuee, 2025-05-05:
No issues with latest avrdude git.
```
mcuee@UbuntuSwift3 ~ $ avrdude -c arduino_as_isp -P ch340 -b 19200  -p m2560 -U read_m2560.hex
Reading 7626 bytes for flash from input file read_m2560.hex
Writing 7626 bytes to flash
Writing | ################################################## | 100% 7.48 s 
Reading | ################################################## | 100% 4.41 s 
7626 bytes of flash verified

Avrdude done.  Thank you.

mcuee@UbuntuSwift3 ~ $ avrdude -c stk500v1 -P ch340 -b 19200  -p m2560 -U flash:v:read_m2560.hex:i
Verifying 7626 bytes of flash against input file read_m2560.hex
Reading | ################################################## | 100% 4.42 s 
7626 bytes of flash verified

Avrdude done.  Thank you.

```

mcuee, 2025-05-05:
No issues using @MCUdude's Mega Core as well.

```
/home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/bin/avrdude -C/home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf -v -patmega2560 -cstk500v1 -P/dev/ttyUSB0 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xf7:m 
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
AVR part              : ATmega2560
Programming modes     : SPM, ISP, HVPP, JTAG
Programmer type       : STK500
Description           : Atmel STK500 v1
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
Vtarget               : 0.0 V
Varef                 : 0.0 V
Oscillator            : Off
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 1E 98 01 (ATmega2560)
Erased chip

Processing -U lock:w:0xff:m
Reading 1 byte for lock from input file 0xff
in 1 section [0, 0]
Writing 1 byte (0xFF) to lock, 1 byte written, 1 verified

Processing -U efuse:w:0xfd:m
Reading 1 byte for efuse from input file 0xfd
in 1 section [0, 0]
Writing 1 byte (0xFD) to efuse, 1 byte written, 1 verified

Processing -U hfuse:w:0xd7:m
Reading 1 byte for hfuse from input file 0xd7
in 1 section [0, 0]
/home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/bin/avrdude -C/home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf -v -patmega2560 -cstk500v1 -P/dev/ttyUSB0 -b19200 -Uflash:w:/home/mcuee/.arduino15/packages/MegaCore/hardware/avr/3.0.3/bootloaders/urboot/atmega2560/watchdog_1_s/autobaud/uart0_rxe0_txe1/led+b7/urboot_atmega2560_pr_ee_ce.hex:i -Ulock:w:0xff:m 
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/.arduino15/packages/MegaCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
Writing 1 byte (0xD7) to hfuse, 1 byte written, 1 verified

Processing -U lfuse:w:0xf7:m
Reading 1 byte for lfuse from input file 0xf7
in 1 section [0, 0]
Writing 1 byte (0xF7) to lfuse, 1 byte written, 1 verified

Avrdude done.  Thank you.
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
AVR part              : ATmega2560
Programming modes     : SPM, ISP, HVPP, JTAG
Programmer type       : STK500
Description           : Atmel STK500 v1
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
Vtarget               : 0.0 V
Varef                 : 0.0 V
Oscillator            : Off
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 1E 98 01 (ATmega2560)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip

Processing -U flash:w:/home/mcuee/.arduino15/packages/MegaCore/hardware/avr/3.0.3/bootloaders/urboot/atmega2560/watchdog_1_s/autobaud/uart0_rxe0_txe1/led+b7/urboot_atmega2560_pr_ee_ce.hex:i
Reading 438 bytes for flash from input file urboot_atmega2560_pr_ee_ce.hex
in 2 sections of [0x3fe00, 0x3ffff]: 2 pages and 74 pad bytes
Writing 438 bytes to flash
Writing | ################################################## | 100% 0.65s
Reading | ################################################## | 100% 0.30s
438 bytes of flash verified

Processing -U lock:w:0xff:m
Reading 1 byte for lock from input file 0xff
in 1 section [0, 0]
Writing 1 byte (0xFF) to lock, 1 byte written, 1 verified

Avrdude done.  Thank you.
```

mcuee, 2025-05-05:
Looks like the issue is with MicroCore -- ATtiny13. I can reproduce the issue. 

No issue if buring the blink ino.
```
/home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/bin/avrdude -C/home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf -v -pattiny13a -cstk500v1 -P/dev/ttyUSB0 -b19200 -Ueeprom:w:/tmp/arduino_build_939525/Blink_t13.ino.eep:i -Uflash:w:/tmp/arduino_build_939525/Blink_t13.ino.hex:i 
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
AVR part              : ATtiny13A
Programming modes     : SPM, ISP, HVSP, debugWIRE
Programmer type       : STK500
Description           : Atmel STK500 v1
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
Vtarget               : 0.0 V
Varef                 : 0.0 V
Oscillator            : Off
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 1E 90 07 (ATtiny13, ATtiny13A)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip

Processing -U eeprom:w:/tmp/arduino_build_939525/Blink_t13.ino.eep:i
Warning: no eeprom data found in Intel Hex file /tmp/arduino_build_939525/Blink_t13.ino.eep
Reading 0 bytes for eeprom from input file Blink_t13.ino.eep
in 0 sections of [0, -1]: 0 pages and 0 pad bytes
Writing 0 bytes to eeprom
Writing | ################################################## | 100% 0.00s
Reading | ################################################## | 100% 0.00s
0 bytes of eeprom verified

Processing -U flash:w:/tmp/arduino_build_939525/Blink_t13.ino.hex:i
Reading 78 bytes for flash from input file Blink_t13.ino.hex
in 1 section [0, 0x4d]: 3 pages and 18 pad bytes
Writing 78 bytes to flash
Writing | ################################################## | 100% 0.23s
Reading | ################################################## | 100% 0.09s
78 bytes of flash verified

Avrdude done.  Thank you.
```

But it will fail to burn the bootloader.
```
/home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/bin/avrdude -C/home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200 -e -Ulock:w:0xff:m -Uhfuse:w:0xfb:m -Ulfuse:w:0b00111010:m 
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/.arduino15/packages/MicroCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
Setting bit clk period: 32.0 us

Error: cannot get into sync
Error: cannot set Parm_STK_SCK_DURATION
Error: unable to open port /dev/ttyUSB0 for programmer stk500v1

Avrdude done.  Thank you.
Error while burning bootloader.
```

mcuee, 2025-05-05:
Same issue with avrdude git. I think the issue is the extra `-B32` which is probably not compatible.

```
mcuee@UbuntuSwift3 ~ $ avrdude -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200 -e -Ulock:w:0xff:m -Uhfuse:w:0xfb:m -Ulfuse:w:0b00111010:m
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
Setting bit clk period: 32.0 us

Error: cannot get into sync
Error: cannot set Parm_STK_SCK_DURATION
Error: unable to open port /dev/ttyUSB0 for programmer stk500v1
A possible candidate serial port is:
  -P /dev/ttyUSB0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

Avrdude done.  Thank you.
```

It will be okay after the removal of `-B32`.

```
mcuee@UbuntuSwift3 ~ $ avrdude -v -pattiny13a -cstk500v1 -P/dev/ttyUSB0 -b19200 -e -Ulock:w:0xff:m -Uhfuse:w:0xfb:m -Ulfuse:w:0b00111010:m
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v1
Setting baud rate     : 19200
AVR part              : ATtiny13A
Programming modes     : SPM, ISP, HVSP, debugWIRE
Programmer type       : STK500
Description           : Atmel STK500 v1
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
Vtarget               : 0.0 V
Varef                 : 0.0 V
Oscillator            : Off
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 1E 90 07 (ATtiny13, ATtiny13A)
Erased chip

Processing -U lock:w:0xff:m
Reading 1 byte for lock from input file 0xff
in 1 section [0, 0]
Writing 1 byte (0xFF) to lock, 1 byte written, 1 verified

Processing -U hfuse:w:0xfb:m
Reading 1 byte for hfuse from input file 0xfb
in 1 section [0, 0]
Writing 1 byte (0xFB) to hfuse, 1 byte written, 1 verified

Processing -U lfuse:w:0b00111010:m
Reading 1 byte for lfuse from input file 0b00111010
in 1 section [0, 0]
Writing 1 byte (0x3A) to lfuse, 1 byte written, 1 verified

Avrdude done.  Thank you.
```

mcuee, 2025-05-05:
@MCUdude and @stefanrueger 

I tend to think this is an issue with MicroCore and not avrdude.
1) Programmer should be `-c arduino_as_isp` to avoid issues with EEPROM.
2) `-B32` is probably not to be used when programming the bootloader.

mcuee, 2025-05-05:
@MCUdude and @stefanrueger 

Hmm, I changed my mind, it seems to be a regression from avrdude 7.2 to avrdude 7.3.
`-B32` works with avrdude 7.2 release but not 7.3 release.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude72 -C ./avrdude72.conf -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200

avrdude72: Version 7.2
           Copyright the AVRDUDE authors;
           see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

           System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude72.conf
           User configuration file is /home/mcuee/.avrduderc
           User configuration file does not exist or is not a regular file, skipping

           Using Port                    : /dev/ttyUSB0
           Using Programmer              : stk500v1
           Overriding Baud Rate          : 19200
           Setting bit clk period        : 32.0
           AVR Part                      : ATtiny13A
           Chip Erase delay              : 4000 us
           RESET disposition             : dedicated
           RETRY pulse                   : SCK
           Serial program mode           : yes
           Parallel program mode         : yes
           Timeout                       : 200
           StabDelay                     : 100
           CmdexeDelay                   : 25
           SyncLoops                     : 32
           PollIndex                     : 3
           PollValue                     : 0x53
           Memory Detail                 :

                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
             flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
             lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
             calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

           Programmer Type : STK500
           Description     : Atmel STK500 version 1.x firmware
           Hardware Version: 2
           Firmware Version: 1.18
           Topcard         : Unknown
           Vtarget         : 0.0 V
           Varef           : 0.0 V
           Oscillator      : Off
           SCK period      : 0.1 us
avrdude72: AVR device initialized and ready to accept instructions
avrdude72: device signature = 0x1e9007 (probably t13a)

avrdude72 done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude73 -C ./avrdude73.conf -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200

avrdude73: Version 7.3
           Copyright the AVRDUDE authors;
           see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

           System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude73.conf
           User configuration file is /home/mcuee/.avrduderc
           User configuration file does not exist or is not a regular file, skipping

           Using port            : /dev/ttyUSB0
           Using programmer      : stk500v1
           Setting baud rate     : 19200
           Setting bit clk period: 32.0 us

avrdude73 stk500_setparm() error: cannot get into sync
avrdude73 stk500_set_sck_period() error: cannot set Parm_STK_SCK_DURATION
avrdude73 main() error: unable to open port /dev/ttyUSB0 for programmer stk500v1
A possible candidate serial port is:
  -P /dev/ttyUSB0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude73 done.  Thank you.
```

mcuee, 2025-05-05:
avrdude issue created.
* https://github.com/avrdudes/avrdude/issues/1978

MCUdude, 2025-05-05:
The simple fix for this issue is to _not_ use the *Arduino as ISP (slow)* programmer option, but rather *Arduino as ISP*.

It appears that the Arduino as ISP has a hardcoded ISP clock speed, and will actually crash or freeze if you try to set a different bit clock speed through Avrdude. The reason why this worked with Avrdude 7.2 or older was that the code that actually sends the bitclock command to the programmer wasn't implemented, so the programmer never got that command that would have caused it to crash.

So older versions of Avrdude had a bug that, in practice, made Avrdude ignore user-specified `-B` bitclock speeds, which clearly was a bug. However, when this was fixed, the Arduino as ISP suddenly didn't like this command because it wasn't implemented.

We'll figure out how Avrdude should deal with this limitation, but the quick fix for this particular core is to remove the `Arduino as ISP (slow)`, as it causes issues, and because the Arduino as ISP has a hard-coded bit clock anyways.

JoergCH, 2025-05-05:
_> The simple fix for this issue is to not use the Arduino as ISP (slow) programmer option, but rather Arduino as ISP._

Are you sure about this? I have always been using "Arduino as ISP".

MCUdude, 2025-05-05:
> Are you sure about this? I have always been using "Arduino as ISP".

@JoergCH you're right! `Arduino as ISP (slow)` won't work, but I've actually forced the `-B32` flag when writing fuses or burning the bootloader. So the issue will hit you regardless. I'll figure out something...

https://github.com/MCUdude/MicroCore/blob/e0bff3cd31c5e0867af24eb76467abd83df4f63c/avr/platform.txt#L152-L156

MCUdude, 2025-05-05:
Issue fixed in 07e999f. A new boards manager release will be available in a few days

commit 07e999fc (2025-05-05): Fix issue where the Arduino as ISP wouldn't program the target device
Closes #163
closed 2025-05-05 by MCUdude